betaflight/betaflight #40: 2.1.4 issue with sbus - self arming

icdark, 2015-11-25:
Hi.
I've just updated from 2.1.3 to 2.1.4. Copied dump all and pasted to new version. After connecting battery it armed itself and started to spin the motors with strange behaviour. Went back to 2.1.3 and it seems to work fine. No idea what happened.

DODO v2, X4R sbus

borisbstyle, 2015-11-25:
I use X4R as well and no issues. I think your serialrx_provider is not correctly set during cli copy paste. I think it is trying to use SPEK1024 :D

set serialrx_provider = SBUS

or set it in configurator config pate.

icdark, 2015-11-25:
That was the first thing I checked. It didn't change from the gui.
I tried again and it worked from the cli.

Nevertheless I think it would be safer if there were some bigger warnings for serialrx_provider ;-)

closed 2015-11-25 by icdark

borisbstyle, 2015-11-25:
It's in the release notes to check all CLI parameters :dancer: 

Also pid_at_min_throttle is different now!
Choice between: OFF, PD, PID