betaflight/betaflight #44: CC3D cut out for no reason..

mattyra, 2015-11-26:
Hello,

I have noticed that the cc3d for no reason shuts the motors down when hovering .....

It's happened a few time now, I'm running the latest Beta flight I flashed the .BIn via open pilot..

I have flashed two cc3d now and same problem but I can't pin point the issue..

I have turned fail safe off to test still the same.

Its like some on hits the disarm button but I can then pull the throttle down and start motors again its really odd...

Matt

borisbstyle, 2015-11-27:
I don't have this issue and many others didn't experience it as well. It must be an ESC or other issue.

mattyra, 2015-11-28:
I dont agree sorry.. the Issue only happens with Betaflight...

if i turn of the disarm motors the issue goes away, its seems as though this is kicking in randomly ?

Thanks

borisbstyle, 2015-11-28:
Yeah but you are the only 1 having this issue :)

maybe misconfiguration.

handsomejackuk, 2015-12-05:
had this issue too.... not sure what caused it i have reflashed 2.1.4 again and  this not happened since...

MrKarver, 2015-12-07:
Hey guy's, saw this and thought I should chime in. Two CC3D's and same problem. going to flash the latest and see if it goes away. 

Just so you know your not alone.

borisbstyle, 2015-12-07:
Did the 2.1.3 version work?

Arakon, 2015-12-07:
I've flown a 2.1.3 copter for ~5 mins without issue (maiden flight, not much of a chance to fly any more).
My older, pre-new-version-numbering betaflight 250 size copter has flown for hours with no issue.
Both are running a banggood cc3d board.

henn1001, 2015-12-07:
No issue here either...

borisbstyle, 2015-12-07:
I really think you guys have ESC issues or something.

MrThrust, 2015-12-09:
had a few similar issues, tracked it down to the 5v source powering the board!
would drop to around 4.6v when trying to throttle up!... 
added dedicated 5v mini bec, problem gone :-)

borisbstyle, 2015-12-11:
So no software issue

closed 2015-12-11 by borisbstyle

mattyra, 2015-12-12:
wow some one is papering over the cracks... i have a dedicated BEC gone back to clean flight and no issues..

Arakon, 2015-12-13:
Try using the non-OP version (that one is really tight on space) and also test with older versions. You'll have to do some narrowing down here, since noone else seems to have these exact problems.

mattyra, 2015-12-13:
Im going to change boards... What the beat fc to get for betaflight

gatmezeiz, 2015-12-16:
I have similar issue (using CC3D) on one of two quads (both have CC3D, the other one runs 2.1.5 without problems). The second one does not start the motors even when I see on the motor tab in cleanflight configurator that the motors should work. This happens since 2.1.5 is flashed. Till that it worked well with 2.1.2. I tried to put back 2.1.2 but did not help. 

Any idea?

mattyra, 2015-12-16:
Borris, 

This topic should be reopened....

Thanks Austen