betaflight/betaflight #45: When saving config CC3D won't reconnect

mattyra, 2015-11-26:
Hi,

When I update settiving in the configuration tab, and the board restart it does not re connect to the GUI I have to also have the battery connected to get the CC3D to be picked up ....

Running latest beta flight...

Matt

Arakon, 2015-11-27:
Not related to BF. CC3D runs a virtual USB port, reboot kills the port, windows doesn't pick that up, doesn't free up the port, so reconnection fails.

As for your battery-first thing, if you flash the .hex from CF Configurator and wipe the OP bootloader that way, this is no longer needed.

borisbstyle, 2015-11-27:
So we can close this one :D

closed 2015-11-27 by borisbstyle

mattyra, 2015-11-28:
Im not running windows, guys.... but i suppose its the same...

Now i have tried every way to flash the cc3d via CF and it just gives NR from Boot Loader....

extraReal, 2016-06-09:
I am having the same exact problem. I'm on the Mac also running the latest release of betaflight (2.7.1). I flashed betaflight and got it to connect to cleanflight with no issue the first time. Did a save / reboot once and it worked. Then I lost connection to cleanflight and had to power cycle the board to get it to reconnect. Once it was reconnected I noticed the settings change didn't take. 

Tried flashing the .hex for cc3d but I also get No response from bootloader. I can get cleanflight to connect with a power cycle of the board but save and reboot doesn't work. Tried several different methods, battery connected, battery not connected. 

Suggest re-opening this issue or I can open a new one, let me know.