betaflight/betaflight #46: Which release for CC3D?

mattfrear, 2015-11-27:
Hello,

Could you add some release notes to the releases page for 2.1.4 as I'm not sure which CC3D one to download. I don't know what BP6 and OPBL and OPBL_BP6 are.

betaflight_CC3D.hex
betaflight_CC3D_BP6.hex
betaflight_CC3D_OPBL.bin
betaflight_CC3D_OPBL_BP6.bin

Also the betaflight_CC3D.bin is missing from the releases page (if that's the vanilla CC3D one I should be downloading.

Thanks

Matt

MarcoH64, 2015-11-27:
The info is out there. I know i read it before. BP6 is for beeper on pin 6.

mattfrear, 2015-11-27:
Ah I see. Thanks, I hadn't figured out the naming convention.

mattfrear, 2015-11-27:
And I guess OPBL is Open Pilot Bin Loader or something, which is what I was looking for.

closed 2015-11-27 by mattfrear

borisbstyle, 2015-11-27:
it cant be more clear.

opbl = open pilot bootloader
bp6 = beeper pin6