betaflight/betaflight #52: I2c gets locked up because of overclocking

universam1, 2015-12-01:
I would love to use Betaflight, unfortunatelly it is not stable. U must admint that I use a DIY Naze32 style board, which runs with cleanflight nicely.
Unfortunatelly, with Betaflight the I2C gets locked after some minute or two. After hours of testing I'm sure of the reason, that it's too much overclocked with MPU6050, BMP180 and HMC 5883L connected.

Would it be possible to intruduce a CLI command for lower I2C speed?

Thank you!

borisbstyle, 2015-12-01:
Your i2c bus is overloaded with all those components. Maybe even the
distances between the components are too big.

I have a list of supported/tested boards. I cant test boards that I dont
have.

I would either remove mag or baro. Both do introduce a lot of jitter and
slow things down.

I prefer to have all static/tested settings where I can guarantee that
everything works as supposed. When people would change settings like this
timing issues might occur and jitter will be increased.

universam1, 2015-12-01:
Thats true, but cleanflight itself introduced a overclock command too, couldn't it be used here as well but be turned on by default?
Wouldn't it be sad that betaflight be a corner product because it won't run under certain conditions?
I think many extend to more or extern sensors so I don't think I'm the only one.

borisbstyle, 2015-12-01:
betaflight is a "proof of concept" for cleanflight not a full product.

If you run "naze32" hardware it will be too slow to run all those sensors
at 1khz speed. Thats what I am saying.

universam1, 2015-12-01:
OK, thanks.

I've managed to get around by deactivating unneeded HW, mag_hardware=1 and baro_hardware=1.

borisbstyle, 2015-12-01:
than your issue is probably something else than i2c overclock.

Does it work with cleanflight?

universam1, 2015-12-01:
Yes it runs nicely with cleanflight, tested runtime more than 4h with all sensors! With Betaflight not more than 30 seconds.

borisbstyle, 2015-12-01:
And when you overclock i2c in cleanflight? What happens than.

universam1, 2015-12-02:
Indeed, with enabled overclocking in CF it works still...
So where is the problem with BF?

Trussroads, 2015-12-02:
Boris is very generous with his work and
I do not think it is appropriate to insult that generosity as you have done here? Boris is not responsible for errors you make while incorrectly using BF and you should be more respectful mr Universam1..
I do not see a fork that you are publicly sharing or have worked tirelessly on to help racers find optimised PID or flight performance or for the general betterment of the sport?
In my own opinion it may be better for you to not use it, as I don't believe you understand the reason for its existence?
Perhaps you can fork CF yourself and develop something for your own bench top use rather than insult other highly respected developers and their work..

Regards,
Scott Chard..
0448411827.

voroshkov, 2015-12-02:
I guess that @universam1 is just trying to point out to a possible issue, which being identified and fixed could make BF better.

borisbstyle, 2015-12-02:
It must be something with baro or mag.

Sorry I cant help you with this as thid hardware is propriatary.

Maybe its the gyro interrupt what doesnt work on you board

universam1, 2015-12-02:
Thanks for the help so far - it wasn't meant to blame anybody anyhow so far!

Just wanted to ask if thats possibility, if not that's an answer.

Thanks

closed 2015-12-02 by universam1