betaflight/betaflight #53: CLI parameters not taken

dropax, 2015-12-03:
More a cosmetic issue as baro functions have a small user base:

baro_noise_lpf, baro_cf_vel and baro_cf_alt seem to be numeric parameter but CLI only accepts ON/OFF inputs. 

Relevant code is located in lines 615-617 of 
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/io/serial_cli.c

Putting code from cleanflight there is tested working.

borisbstyle, 2015-12-03:
Thanks for noticing. Will fix that

borisbstyle, 2015-12-04:
Cf also only allowes 0 and 1 btw. So I guess ON/OFF is fine

dropax, 2015-12-04:
Defaults are 0.600, 0.985 and 0.965 and listed as FLOAT inCLI.md so I guess they are meant to be in a range 0-1 and not just on/off. 
With the code from cleanflight patched in serial_cli.c everything is fine, you can feed the commandline with float values and they are prompted back as entered. ;-)

borisbstyle, 2015-12-04:
yes I saw that just after my comment. I already fixed it

dropax, 2015-12-04:
*thumbsup*

closed 2015-12-04 by dropax