betaflight/betaflight #54: Rev 6 Auto level  Gyro saturation

mattyra, 2015-12-04:
Hi,

Just a heads up the rev6 is having issues holding level, it always tilts off to the left.... its the same on 3 boards now...

all calibrations done..... its the same from take of to switching back from Arco mode its goes into a huge left roll....

Matt

borisbstyle, 2015-12-05:
I think something isnt calibrated properly in your case

mattyra, 2015-12-05:
such as ???

readerror67, 2015-12-05:
Sounds like a bunch of user error taking place, reference:
https://github.com/cleanflight/cleanflight/issues/1321#issuecomment-152046620

mattyra, 2015-12-05:
sounds to me like https://github.com/readerror67 https://github.com/cleanflight/cleanflight/issues/1467 

now if there is a problem with calibration can some one let me know...  now i have over 16 250/200 sat on my wall and the issue remains only on the rev6 board, now even odder after some time the issue clears and it levels again almost like its hanging..

Now if I'm wrong then try to help as i am only trying to highlight the fact that there is a problem it may be useful to help others...

Thanks

mattyra, 2015-12-05:
https://github.com/cleanflight/cleanflight/issues/1415

jtp755, 2015-12-05:
@mattyra Wish i had a rev6 to help test with but could you post your quad specs? I think you have definitely done your research and looking in to the problem but could you also quickly recap the steps/things you've done for calibration and set up as well as the steps to reproduce?

When you flash the firmware...are you doing a full chip erase and not restoring any CLI dumps? If you are I would say to start there and NOT do any restores....just basic settings for operation as to remove as many variables as possible.

borisbstyle, 2015-12-07:
Can you please continue with rev6 issues in cleanflight issues. You should also try CF 1.11 latest pre-release to compare.

closed 2015-12-07 by borisbstyle