betaflight/betaflight #55: 60-80% Spool up on Arm with Air Mode enabled - Mini Naze32 boards

cheetah2k, 2015-12-06:
Great way to destroy a quad.  Using Air Mode, arming the quad and watching it spool up to between 60-80% throttle without anything you can do, except disarm.

Without Air Mode enabled, zero issues, except for the quad rolling by itself on low throttle decents..

jaapp, 2015-12-06:
Was your quad on the ground while it was in Air Mode?

borisbstyle, 2015-12-06:
This is now addressed in the latest code. Please try latest hex :D. Remove props prior testing :D

closed 2015-12-06 by borisbstyle

cheetah2k, 2015-12-06:
Thanks @borisbstyle   

@jaapp this issue took me by surprise.  I had set up 2 quads with SPRF3 boards (bench tested with zero issues and flown perfect), then setup an identical quad but with mini naze32 and assumed it would also be perfect, placed it on level ground, armed in Horizon mode, and it just took off by itself...  It was 20m+ high before I realised the only thing I could do was disarm :(  I will try the revised hex this afternoon and report. 

Learnt a major lesson - every quad is different.  benchtest always.

cheetah2k, 2015-12-06:
@borisbstyle I have a spare Multiwii Hero32 FC if you need for testing.  PM me your details and I'll post it to you (willing to sacrifice it for the Scarab Knife community) and for the good of Betaflight

borisbstyle, 2015-12-06:
If you want try it out. I did find a minor bug which could lead to something like that. I am confident it is solved.

le3dmax, 2016-02-22:
I'm having this issue again with the 2.4.0 and 2.4.1, I can't arm with the air mode on, if I do it, it spool up, or it flips over instantly... any idea why?