betaflight/betaflight #57: GPS configuration

MarceloFarinon, 2015-12-06:
With the version 2.14 I cannot change the GPS configuration to UBLOX. Do you have removed the support for GPS?

borisbstyle, 2015-12-07:
Cli command has changed:
set gps_provider = UBLOX

closed 2015-12-07 by borisbstyle

MarceloFarinon, 2015-12-07:
Sorry, but "command invalid".
By the way, I didn´t find this command using "dump" on CLI.

Cli command has changed:
set gps_provider = UBLOX

borisbstyle, 2015-12-07:
maybe gps is disabled on your board.

Some boards dont have it due to memory space like cc3d for example

Schalonsus, 2015-12-15:
Does not work on Naze32 rev5 Acro and Flip32+ 10DOF. What a bummer.

MarceloFarinon, 2015-12-15:
I installed Cleanflight latest version 1.10 and 1.11 versions and it´s working fine... so I believe there is something wrong with Betaflight version.

Schalonsus, 2015-12-15:
There is nothing wrong i think. It is disabled to make place for the gtune code.
In a german forum someone is willing to recode the latest betaflight firmware to activate GPS and deactivate gtune.
So you can use the standard betaflight firmware to tune your copter and after that flash the recoded one to use GPS. He said he will try to code this this evening if he has time for it.
Will report back if there are any news.

dakingen, 2015-12-22:
Not working on Naze32 Rev6 Acro. (working in Cleanflight)
Boris, can you confirm that gps is disabled in Betaflight or is this an issue?

borisbstyle, 2015-12-22:
@dakingen 
GPS is disabled on Naze targets to make space available for gtune

https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/target/NAZE/target.h#L172

Schalonsus, 2015-12-22:
Tried enabling GPS und disabling gtune in target.h but gps only works for 20 seconds and then does not work for 20 seconds. Any hint on this term?

fellebomp, 2016-10-20:
Version 3.0.0 (stable), on Naze32 FULL: Go to target.h file in \src\main\target\NAZE and search for the line where GPS is disabled in the sourcecode: //#undef GPS

For having enough flash memory on the STM32 controller I disabled the BLACKBOX logger and enabled the GPS:

#define GPS
#undef BLACKBOX

Seems to work fine! But I will do some additional testing...

mikeller, 2016-10-20:
@fellebomp: If you are interested in getting better GPS support, you might want to try iNav: https://github.com/iNavFlight/inav

fellebomp, 2016-10-20:
Thanks for this hint! 
My idea was to have betaflight installed on the naze to get it's really good flight characteristics for my race kwad - much better than cleanflight AND only show the most important flight information on OSD like battery voltage, RSSI, altitude, speed and distance to home. For this purpose this configuration seems to work pretty well - so far!

VoicOfReason, 2017-08-24:
I can confirm that GPS Smart port Frsky receiver works. Does anyone know if the F7 boards support GPS on Betaflight? I am not interested in RTH or anything, just getting speed, location, etc.

mikeller, 2017-08-25:
@VoicOfReason: Yes they do:

https://github.com/betaflight/betaflight/blob/master/src/main/target/common_fc_pre.h#L137

Removal of GPS from F3 is due to flash space limitation, and F7 is not affected (yet).

VoicOfReason, 2017-09-28:
OK. How about F3 boards? Are they affected, as well, or do they have enough memory for GPS?

Schalonsus, 2017-09-28:
You should better use iNav for gps assistance

VoicOfReason, 2017-09-28:
I appreciate your advice. I am already using iNav on another quad. But, I
want to use a GPS on Betaflight in order to JUST get GPS coords. So, back
to my question: will F4 work? Thanks.

jflyper, 2017-09-29:
> will F4 work?

F4 does have GPS code in most, if not all, targets.

For further assistance, please consider using other user support options such as [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

Thank you in advance.