betaflight/betaflight #58: Brushed motor support does not do 100% duty cycle

LukeRitchie, 2015-12-09:
With several 32bit brushed controllers out now and that they fly so much better on betaflight I was thinking maybe this issue could get looked at? I'm no coder but it would seem there should be a simple hack to fix the issue. Maybe it would not require much time to test out at least?   

I base this "simple to fix" theory on Geebles testing. Here is his comment from cleanflight 

"With the current brushed motor support in cleanflight, it appears to just reduce the PWM minimum value only (so that when "Minimum Throttle" is set to 1000, PWM is set to 0%) and the top end is unaffected, meaning that when you set "Maximum Throttle" to 2000, what is actually happening is the PWM is 'on' for '2000 time' and then 'off' for about '250-500' time (this applied to framing the PWM for ESCs, but for brushed it means that full throttle isn't actually 100% PWM duty.

As a quick hack I set "Maximum Throttle" to 2500, which according to the scope shows 100% duty cycle (hooray) however there is some weirdness in the PID controllers because when at full or very high throttle the copter will lean to one side (the acc/gyro is reading this but is ok with it, so the PID controller is making it lean).. reducing back to 2000 and full throttle is vertical as expected."

Thanks! If anyone has any idea how the actual code works for this it would be great.

borisbstyle, 2015-12-09:
Unfortunatelly I have no brushed motors / setups to do testing with it

LukeRitchie, 2015-12-10:
If you are willing to have a look at it I'm sure I could organize some micro gear for you to test?

borisbstyle, 2015-12-11:
set max_throttle = 2500 should solve it I heared

Try that. If that doesnt work this issue should belong to Cleanflight issue tracker

commit 5d5fd81b (2015-09-11): Fix bug where PPM rx stops working on sparky or CC3D when using motor
PWM rate > 500 (brushed motor mode).

Fix is the same as for OneShot since both brushed motors and oneshot use
an 8mhz timer.

Fixes #58
closed 2015-12-11 by borisbstyle