betaflight/betaflight #60: Motors spinning at full power on reboot during conf ... - Betaflight 2.1.4 & 2.1.5 RC / IRC Vortex (Naze32 hex)

TiTidom-RC, 2015-12-10:
Hi Boris, 

Conf : IRC Vortex / OSD 1.0.0.21 / Cleanflight configurator 1.1.0 / Betaflight 2.1.4 & 2.1.5 RCx 

[OFF]
First of all, thank's for your hard work, i'm flying with betaflight since the first release and it's ever better on each release :)
[/ OFF]

Since 2.1.4, i have a "little" issue during config on a vortex (on this quad, we need to plug lipo to be able to access usb, that's why i have an issue...)

If i activate arming thru an AUX (aux2 in my config), after that, on each "save and reboot" (config tab for example, in the CLI too), the motors are spinning (with almost full power) themself for about 1 sec before stopping :(

If a use arming through stick, i have not this issue, but as soon as i return to arming thru AUX, the issue returns...

of course, I do the conf without props, but it's always surprising when engins start alone at full power !!

I reverted back to 2.1.3, and have no problem (no motor spinning on reboot) even with arming thru aux.

If you can do something for this issue :)

Olivier.

borisbstyle, 2015-12-11:
This is an old issue and reported somewhere in cleanflight issues. Somehow only vortex has this problem.

TiTidom-RC, 2015-12-11:
Yes, i knew this old issue in Cleanflight. What is stange is why this issue appear and disappear from one realease to another without specific change in the code ? 

For example : in betaflight 2.1.3, no issue with motor spinning... in 2.1.4 and 2.1.5 issue with motor spinning...

Strange...
@++

borisbstyle, 2015-12-11:
It doesnt have to do with releases. It is something what needs certain
scenarios to be triggered.
Even I can have no issue like 99 times and 100th time it will happen. It is
timing/scenario related.
Somebody should investigate that once, but as only happens to vortex there
are not much folks who spend time on this.

ChaosPower, 2015-12-22:
Just purchased a vortex. Will help investigate this.

sshirokov, 2016-01-01:
I just ran into this issue with a Naze32 Rev6 (no mag) with both 'Betaflight 2.2.0 RC2' and 'Betaflight 2.2.0 RC2' using a Turnigy 9x 8ch TX/RX pair.

I didn't wait long enough to see if the motors spin down after a timeout, but toggling the arm switch seemed to only change the RPM, not star/stop the motors. I confirmed in the transmitter tab that the aux channel levels were correct.

Let me know if you want any more details, I'm pretty certain I can reproduce it trivially.

borisbstyle, 2016-01-01:
For stopping the motors your min_comand has to be configured low enough for
that it that is the case.

borisbstyle, 2016-01-20:
I was not able to reproduce this issue anymore. I will close the issue. Please reopen if it is still there.

It is anyway always advisable to do changes without props

closed 2016-01-20 by borisbstyle

ChaosPower, 2016-01-21:
@borisbstyle I was able to replicate this issue (different from @sshirokov's case) and is common scenario for my Immersion RC Vortex 285 Fusion FC Gen1. 

The cause is most probably power source. The LiPo should be attached to the drone (!!no props please!!) before it can be configured / discovered. When Cleanflight configurator saves and reboots, a signal seems to be sent to the esc provided by some software init or hardware. 

However occasionally it does not happen, and when it does not happen Serial_Rx is off at Ports tab. When I turn it on again + save reboot then it spins sometimes in full power and sometimes min throttle.

My hunch is probably one of the init functions.

serialInit
mixerInit
rxInit
failsafeInit (?)

I think the configurator is at fault as well. The signal to the ESCs does not seem to clear up / retain value when it reboots.

Currently finding time to debug the hardware and software. Will upload a video.

universam1, 2016-01-21:
Are you running Oneshot?
I can trigger that by unplugging an powered up ESC - a short pulse somehow
reads the ESC as a command. If thats your root cause then it's most
probably an Oneshoot design issue, not the FW

ChaosPower, 2016-01-23:
Probably is OneShot. But why does it only happen when Serial RX is turned on? Seen an issue with OneShot glitches (PPM) probably that one is linked to what I am experiencing. Will open an issue at BLHeli repo when I confirmed this.  Thanks @universam1