betaflight/betaflight #64: Cut out on cc3d again

mattyra, 2015-12-13:
Borris...

Im giving up on the cc3d .... 

The motors keep cutting at random intervals i have run the fc via a independant battery i have run different cc3d board.... But still it just randomly cuts i have this on a shen tweeker and thug 🐖....

I flashed the latest version and when it is going its awsome... Now just for clarity if it cuts out when im hight enough i can drop the throttel stick and re engage the motors and of shr goes..

Oddly if i chane yo air mode and then back some times it to drops out for no reason...

Like i say i have run a dedictaed bec so now totaly baffeled....

I have also strapped the xt60 connectors and run a voltage record and there are no drops... It really does seem software....

One more thing on my last flash of the cc3d i noticed that the craft was laggy on the live screen... So i checked with one running a naze and all was fine... I then reflashed libra pilot to the board took it out and every thing was fine.....

Any way keep up the good work... Can i just ask one qestion there are loads of fc board on the market what would be your goto fc board ???

handsomejackuk, 2015-12-13:
hmmm same thing happening here on cc3d atom..Random dropouts... Josh recommended altering rxfail to 3h... i have done this and give it a try tomorrow

handsomejackuk, 2015-12-13:
i thought it was losing RC command... I did a blackbox log a couple of weeks ago, and there was a spike in the rc command inputs which led me to believe i maybe had a loose ppm signal wire, i resoldered. my ppm, i am using a flysky receiver 8 channel that has been proved to be real reliable so far.. i also got a pwm ppm convertor this i have used on my naze32 with no problem. and i am not flying out of range as we are only talking here a few metres line of site... happen on fast yaws, and even on just a hover.. so it seems its not just me  hope you can help Boris or suggest any further tests i can do....this happened before air mode on previous versions of betaflight too.. thanks Al

borisbstyle, 2015-12-14:
Maybe but just maybe there is huge amount of jitter on CC3D. From what you are saying there @handsomejackuk indicates something like that.
try "set rc_smoothing = off" and see if that solves the spiking RC commands.

handsomejackuk, 2015-12-15:
thanks Boris... i will try that and report back when we get some dry weather her to get out to fly.... Keep up the great work,,,

jrumbelow, 2015-12-16:
Hi first off all Boris. Your firmware is superb had it on naze32 and completely transform my craft, recently broke the naze and moved to a backup CC3D board. And can confirm like handsomejack I have motor shut down when enabling air mode after I do a loop or roll. Ive bought another naze but thought Id share for the ongoing development. Super work once again Boris keep it up.

mattyra, 2015-12-16:
#44 

I Noticed that when i disabled the motor arm switch it all seemed okay.. i also changed the cut of time to 0secs

borisbstyle, 2015-12-16:
Thanks for sharing all this, but to me the issue doesnt seem to be in the code. All other targets just work and quite a few folks I know dont have the issue on their CC3D

Could you also try 2.1.6 to see if the problem is still there on your boards.

marbalon, 2015-12-17:
I have a question for people with this problems. Could you check loop time in GUI when you reset board byt hardware or software few times ? It is constant 1000us with +/-5 ? I found that my MPU-6000 sometimes start with bad initialization and works with 8KHz without sample divider so gyro sync does not work properly. Solution for me is bigger delays in mpu6000AccAndGyroInit() 15ms instead 1ms.

borisbstyle, 2015-12-18:
@marbalon and others
Can you verify that the issue is the low delay?

marbalon, 2015-12-18:
I can confirm this happens on my custom board with MPU-6000, but problem occurs occasionally. But with more delays in init fix the problem in 100%. It is easy to see this in loop time info but I made some dumps using logic analyzer.

znarf7, 2015-12-19:
After I had some odd problems with my copter today aswell (CC3D running 2.1.6) i fund this thread and had a look at the loop time. It is as you described the problem @marbalon the loop time sometimes spikes up to something over 2000 and is not stable in general.
how did you implement the bigger delays in mpu6000AccAndGyroInit()? i´d like to try that.

marbalon, 2015-12-19:
I just put this delays in code and compile. But the problem is that if bad initialization occurs you shoul see loop time is less then 1000us.

LeChuck71, 2015-12-19:
hm,
after connecting with the latest version of betaflight my cycle time is between 995 ... 10200
I am not a bit concerned if this is the right firmware for my cc3d. Although I am looking forward using it
Is it save to do so?

henn1001, 2015-12-19:
The looptime fluctuations are related to the vusb port.. try again with a ftdi on either of the 2 usarts.

LeChuck71, 2015-12-19:
Ah I hoped it was something that I did "wrong" ;)

LeChuck71, 2015-12-19:
Just connected via FTDI and its between 1008...994

handsomejackuk, 2015-12-21:
is the vusb port the onboard usb one...? as i dont use that, I only use main port for flashing and running a bluetooth adaptor... as my usb port is hidden inside my zmr lowrider frame...(main port next to flexi port)... would there be any advantage / disadvantage to disabling the main usb connect on an atom... sorry not got back to testing any more as been a bit il with flu.. back again soon....

henn1001, 2015-12-21:
Yes i meant the onboard usb..  it should not matter as long as there is nothing connected to it.

MrKarver, 2015-12-22:
More info, I tried with and without OPBL on Betaflight 2.1.6 and in both instances the board would crash (all motors full stop) in flight. Flashed back to CF 1.11.0 and 8 batteries with no cut outs. Are there any suggestions as to something I can try to get it stable on CC3D? I've tried two different boards and same thing. 

I'm thinking of ordering Motolab Tornado's to replace CC3d's thoughts? I see Boris flies the Tornado would this be my best bet for a stable Betaflight platform?

mattyra, 2015-12-27:
I think If beta is not going to support CC3D they should just be more open.... After all buisness is buisness

borisbstyle, 2015-12-28:
@MrKarver 
Can you try on Cleanflight 1.12 with gyro_sync enabled.

@henn1001 
Is there an issue with CC3D? Your works fine right?

MrKarver, 2015-12-29:
@borisbstyle 
I flashed 1.12 and enabled gyro_sync = ON the cycle time was hovering around 1000. That resulted in about 3 times more "cutouts" and it wasn't flyable. I then set gyro_sync_denom = 2 and it was stable. Burned up 4 batteries testing with no trouble at all over many USB connects/disconnects/reboots/config changes. Seemed great.

Do we just have some bad samples of the CC3D that wont run 1 to 1 with the gyro? I'm having the same result with 2 different CC3D's both from good suppliers.

1.12 with gyro_sync enabled feels very nice on the sticks!

henn1001, 2015-12-29:
Haven´t flown that much lately because of the weather but i did 5 flights with airmode and at least 30 with the version pre air mode and never noticed anything strange...

cyberbaloo77, 2015-12-29:
Hi i also have an issue with the bf 2.1.5 and 2.1.6 .
I had 2.1.3 on all of my quads without any problem i have one cc3d and one atom. I updated from 2.1.3 to 2.1.5 on one of my quads and setet and got everything to work except the motors no matter what i do they want spin. i tried my second quad with 2.1.3 there were no problems. i flashed it with 2.1.5 and the same thing there they wont spin not with the tx or the program. 
Have looked around and my english aint to good and i have never entered anything in the cli so i dont know how to do things there.
And i cant find anything about this problem.
Have a hard time thinking its only me this is happening to. and on 2 of my quads so right now i wont flash my other ones before i can get these 2 to work.

I hope someone here can help me.

Best Regards 
Morgan

MrKarver, 2015-12-29:
@borisbstyle 
Just for fun I flashed another CC3D with current Betaflight. So this is controller is number 3. Same thing flies for a bit then randomly cuts out. much better than the 1.12 with gyro_sync_denom set to 1 that would crash every 20 sec or so but still at least once a battery. 

@cyberbaloo77 
I'm sorry to hear your having issues but it sounds like your having a different problem. The issue we are talking about here is that the flight controller will run the motors for a bit and then shut them off.

borisbstyle, 2015-12-29:
Than there must be an issue with cc3d and interrupt stuff of it if you
experience the same on cf 1.12 when you enable it.

MrKarver, 2015-12-29:
In 1.12 if I set gyro_sync_denom = 2 it works fine.

cyberbaloo77, 2015-12-29:
Ok i manage to fix my problem somehow :) So now i dont going to disturb u anymore :)

borisbstyle, 2015-12-29:
And in betaflight when you put gyro_lpf to off?

LeChuck71, 2015-12-29:
Flew just 1 pack on the latest Betaflight on a cc3d board with no issues at all. First I thought its just a problem in "Air" mode. Actually i flew like 6 flights so far and everything went fine. No expert here but if i can help with some infos or settings i am glad to help.

MrKarver, 2015-12-30:
@borisbstyle 
I had to add a pic for this one. I flashed to 2.2.0 and the result is I get ACC and Gyro data in GUI, receiver inputs react as expected in GUI but when I arm (either with an AUX ch or using yaw right) 3 motors go to full throttle and one stays at idle. As you can see the throttle on the radio is at idle. Tried both CC3D's and exact same result. Am I missing something? Also nothing changed setting gyro_lpf = OFF. Apart from that nothing else seems unusual. Definitely not flyable though. 
[image]

borisbstyle, 2015-12-30:
On cc3d you need to reset all settings on any transition between 2.2.0 and
other firmwares. Maybe thats why strange thing happened.

MrKarver, 2015-12-30:
I am using Cleanflight Configurator to flash via FTDI  and I am selecting "Full chip erase" on the flash utility screen. That should cover it right?

MrKarver, 2015-12-30:
@borisbstyle 
I just flashed a Naze32 with 2.2.0 and under my setup I got the same result as the CC3D in the above picture. Three motors at full throttle and one at idle. I don't think there is anything special about my setup ezuhf with PPM. All the motors work correctly when controlled in the GUI. The frame reacts correctly in the setup screen. The receiver page looks good; all the inputs are reading correctly.

catchthewind, 2015-12-30:
CC3D motors at full throttle is being discussed here https://github.com/borisbstyle/betaflight/issues/88

MrKarver, 2015-12-31:
@borisbstyle 
2.2.0 RC2 6 batteries in and no cut outs with gyro_lpf to off.

catchthewind, 2015-12-31:
CC3D on a tricopter 2.2.0 RC2 - 3 batteries and no cut out. Everything OK.

maddust, 2016-01-01:
I'm having the same cut outs issues with every version above 2.1.3 I'm currently testing 2.2.0 rc2 seems better but occasionally I'm still getting cutouts with Gyro_lpf=off . also have anyone experienced spikes on RC command when 2khz is enabled?.  in my case I think is related to rc commands sometimes I see some glitches in the rc command probably causing the quad to disarm in mid air :(.

borisbstyle, 2016-01-01:
spike in rc command is usually skipped loop by blackbox. Blackbox than
writes the 2 values in same loop. Look for 7loop counter in blackbox to
verify

nikybiasion, 2016-01-02:
Hello, i see that when i power on the board, the looptime is between 488 and 501, but sometimes it start to fluctuate between 480 and 560, seems when i switch to air mode, or when i disarm and change pid controller

borisbstyle, 2016-01-02:
Thats normal cause CC3D uses VCP to connect with configurator. Thats not
the case when you fly

borisbstyle, 2016-01-04:
so is there a CC3D issue or not?

If issue exists even on cleanflight with gyro sync activated please report it there too. It is still questionable weather there is a real issue or not?

nikybiasion, 2016-01-04:
I've resoldered the esc to fc and motor and the issue seems disappeared. But i've only test with one lipo.

Rhoxa, 2016-01-06:
I just flashed my CC3D with 2.2.0 RC2.  I've run about 20 batteries through it on Open Pilot without issue.  Never tried regular CleanFlight.  I got everything up and running quick, but my first hover quickly ended in a cut out.  Tried again today and I had two cut outs in about 2 minutes of hovering.  I'm using everything from MyRCMart: CC3D, Little Bees 20A, etc.  TX and RX are a Taranis and D4R.  Also using most stock settings.  I haven't touched sync, filters, or anything like that.  

To get it working again, I think I just have to let it sit for a second.  I tried dis-arming, re-arming, and changing modes quickly, but nothing worked except waiting about 5 or 10 seconds.  

I'll try some settings changes and report back with details.

destbr, 2016-01-08:
Hi Boris,

I tried with gyro lpf = OFF and the FC disarmed just after armed (BF 2.2.0 Final)
It's a CC3D atom

I confirm that Cleanflight 1.10 doesn't have this issue. I haven't tried newest CF because I can't flash .hex files.
I have 2 CC3D with this problem.

There is a previous BF version that this issue does not happen?

nikybiasion, 2016-01-08:
6 lipo and no issues for me after  i've resoldered esc to fc
cc3d from myrcmart
gyro_lpf = off dterm_cuthz=0
acc|baro|mag_hardware = 1
receiver x4r-sb
littlebee 20a

destbr, 2016-01-08:
@nikybiasion

I'll try these configurations and test. Thanks!
One of my CC3D is from myrcmart too.

Rhoxa, 2016-01-08:
Tried those above configs over two batteries hovering:
I lost horizon and acro modes.  Only Air mode is available.  Not sure if this is normal?  I was assuming i t was from the hardware settings changes. 
I have even worse cut outs now.  I'm going to reflash and reset everything and see what happened.

borisbstyle, 2016-01-08:
@nikybiasion 
You should try 1.12 release cleanflight. I do recommend to be cautious with older cleanflight releases due to many failsafe bugs

@Rhoxa 
When you disable accelerometer of course level modes dissapear

destbr, 2016-01-09:
I just made some bench tests (with quad in my hands). With gyro lpf = off the motors spins up suddenly and slows down again in a short period of time (frequently).
The looptime is around 500 (2kz), but I still get cut outs.
I checked blackbox and there is no spikes in any graphs

I don't wanna rollback to cleanflight, nor librepilot ;-(

Rhoxa, 2016-01-12:
Solved my problem.  I did a few things at once in a last ditch effort.  First I got rid of my 5V BEC and put in a new PBD with 5V and 12V on it, this of course involved re-soldering my ESCs as mentioned above.  Then I erased everything and re-flashed betaflight.  I sat at the TX graph screen and noticed once in a while one my Aux channels was spiking randomly at about the same frequency I was having my drop outs before.  This same aux channel was the switch I used for arming and it was spiking out of the range I used for arming.  I expanded my arming range to be pretty big (~1400 to max).  

Hovered two batteries with no drop outs so far.  Now I just need to figure out the spike I guess.

borisbstyle, 2016-01-12:
Ok guys so far no issues discovered. I have plenty of good CC3D feedback and all issues seem user related.
I can understand when few users have simillar issue it is easy to think it is software related.
Running cleanflight has also different frequency/looptime so the issue could be harder to reproduce.

I actually have 5 quads and one of the quads had occassional drop outs out of sky. One of the ESC wires was touching some of the FC components and that ESC was freaking out. Moving the wire from FC simply solved the issue. 
The bottom line is that dropouts out of sky are in 99,9% of cases always hardware failure.

closed 2016-01-12 by borisbstyle

MrKarver, 2016-01-14:
Sorry for the late reply. I was away for holidays. I have moved away from the CC3d's to SP Racing flight controllers. I loaded 2.3.0 to day and I am still seeing input command spikes. Which when using a switch for arming which I do for 3d setup. I get cut outs. Like I said I have moved on and I am not really looking back. I can further test if there is anything you would like me to try.

edusteinhorst, 2016-01-31:
@MrKarver @borisbstyle 

I have the same issue with CF 1.12-RC1 and a CC3D board, sudden cut outs. I can confirm that I only get this behaviour with gyro_sync=on. I tested it several times switching between on and off, and it's consistent. The motors usually cut out under 2 minutes when gyro_sync=on. Tried both gyro_lpf=188Hz and 42Hz, and both produced cut outs when gyro_sync=on. 

@marbalon 

I tried increasing the delays in mpu6000AccAndGyroInit() from 1us to 15us (or did you really mean ms?) but that didn't solve the cut outs. Has it worked for you?

edusteinhorst, 2016-01-31:
A quick update: using gyro_sync on with gyro_denom=2 seems to fix or at least drastically reduce the cut outs. Haven't had one yet, but will keep testing.

edusteinhorst, 2016-02-01:
@borisbstyle @MrKarver @Rhoxa 

Ok, so this is really an issue with the aux channels glitching, which disarms the copter if you use an aux switch for arming. The glitch only happens with gyro_sync=on and gyro_sync_denom=1. Otherwise, no glitch and no disarming.

[image]

The glitch takes the channel out of range and disarms it. 

@borisbstyle 

I tried rc_smoothing=off, but no effect.

Any ideas? CPU overload at 1KHz?

borisbstyle, 2016-02-01:
that is nothing new. CC3D has a timer clash on PPM

see this issue. It is waiting for someone to be fixed
https://github.com/borisbstyle/betaflight/issues/127

destbr, 2016-02-01:
As I have 2 CC3Ds with this issue and I use swtich for arming, I'll try to change settings to arming via stick and test!

Thanks!!!!

edusteinhorst, 2016-02-01:
@borisbstyle 

Thanks for the heads up!

destbr, 2016-02-04:
I changed from switch arming to stick arming (+ idle up for airmode) with success. 2 lipos with no disarming!

Thanks!!!

[5 more replies]