betaflight/betaflight #66: balancer feature

universam1, 2015-12-15:
What about a balancer strobe helper function, since BF high gyro rate enables new possibilties!?
Thinking of a function like a stroboscope to balance prop and motors by BF. 
Something like this: https://www.youtube.com/watch?time_continue=1&v=avJG8Bd0Tx0

Idea:
-white led that BF triggers on gyro peak and fires always when peak points to fc / copter center.
-BF runs each motor individualy (via motor tab in configurator) given that BF knows motor position anyway
-samples over some ms the peaks to get rpm and find peaks that are in angle of the run motor
-delay trigger of LED with loop time to be phase correct
-given motor runs @ 1000rpm and lpf disabled, @1khz gyro the resolution would be excellent 6° ( 1000rpm / 60s  / 1khz *360°)

Maybe a crazy idea, but it could work and would be a cool feature :-)

borisbstyle, 2015-12-15:
@universam1 That looks very promising, although I am still processing this information of how exactly it works, but I would love to try it.

universam1, 2015-12-15:
I've contributed to MWII code, but 32bit is new to me so besides of a dirty hack I can hardly contribute currently. But any good hooking place in your loop to start at?

borisbstyle, 2015-12-15:
@universam1 
The code only supports 1khz now. 8khz is impossible with our current hardware. F3 targets can reach up to 2-4khz. You can achive those speeds with putting gyro into 8khz mode and configuring sample drop divider.
The control loop runs at 1khz, but pre-loop tasks and checks can run more often. I am just not sure of how precise your timing has to be and how much jitter is allowed.

@digitalentity has been working on a task scheduler, which removes the superloop entirely and gives more timing freedom for different things. I will soon work to integrate it into the code. That should make task assigning much easier.
I am planning to start with that soon. The code works, but just having some jitter challenges.

universam1, 2015-12-15:
Thank you, valuable input!

I will start with a hook in the control loop that is nicely in sync with the gyro data and see what results I get. 
Not sure how good the statistics need to be to get some usefull trigger.

Since the osciallations are on X an Y axis, I wonder though if gyro or even acc data would give better results?

universam1, 2015-12-16:
So here is the result from my quick hack of BF:

here you see the very first run, nothing optimized. You may recognize the tape I put on this actually quite good balanced little brushless. In video it is harder to see but the tape as unbalancing is quite good detected.

https://youtu.be/A0ZprLq_JGM

As I said nothing tuned. I realize the strobe must be only some micro second short.
I trigger on peaks, I'm sure this can be done way better.

But as a proof of concept it works with betaflight :)

universam1, 2015-12-17:
Now tried with a real copter.
This time the algorithm is not to detect peak but the zero crossing, since the imbalance is kind of a sine wave.

https://www.youtube.com/watch?v=Mhtr1su64hg

lkaino, 2015-12-17:
Nice. So you light the led when the acc value crosses zero (rising or falling)? Are you trying to position the heavy side away from the FC or is it just coincidence?

universam1, 2015-12-17:
Only the rising slope. This run was with gyro, although with acc the results are similar.
Its realted to the axis that is monitored, so it is 90° phase shift from Y-axis that points to the 12 o'clock of the copter. May be a convenient point.

universam1, 2015-12-18:
@borisbstyle Do you see a possibility to phase correct trigger an output? The port must be triggered timing critical with zero delay, something that could be done with timer or interrupt?

Is there a free timer or routine like that is used for oneshot that could fit?

borisbstyle, 2015-12-18:
@universam1 
Oneshot125 is triggered in main loop. There is no timer routine for that.

How much delay is allowed? To get the minimum possible delay you would have to set the system in some kind of "balancer state" as this is not a flying operation a lot of code and tasking can be disabled and you will even be able to get much much higher speeds. Motor update for oneshot125 is limited to 4khz of course

universam1, 2015-12-18:
The delay would not be an issues when it is stable which we could compensate. Only importance is being phase correct which means it should happen in the exact as possble time. 
You see in the video that the trigger is not always at the same time, so it optically moves. I guess we should be in time with ~300us. Thats why I think of a timer interrupt.

My guess is that most accuracy would be to measure the RPM to have a precise timing metro and set a timer interrupt on that. Then we phase shift the metro fitting to the peak or zero crossing.and we are done.

borisbstyle, 2015-12-18:
Gyro is very exact....every 1ms. Not sure exactly how/wherr do you get
300us?

lkaino, 2015-12-18:
From the video it seems that sometimes the light is about 20 degrees off. At 1000rpm that's 3,5 ms. Could there be some harmonic frequencies it's detecting? 

How fast was the motor spinning there?

universam1, 2015-12-18:
The resolution of 1ms is not fast enough to catch always exactly the very
same position of the motor. So some smoothing / averaging thus prediction
of point 0 would help to catch that point. AFAIK there is enough runtime
between the gyro updates, and I predict that we should have a tick of 300us
to get a better result.
On AVR it is quite easy to create a precise timed interrupt, so to run
independently of the gyro update, how about here? Would it mess too much
with the current code?

universam1, 2015-12-18:
@lkaino  good point, I have no idea how fast it runs. I guess way more than 1000rpm.
Would be great if someone else tries that, take my fork.

lkaino, 2015-12-18:
Measure the time between detections using micros() and calculate rpm from that. Put it to debug variable and see from sensors tab how constant values are you getting. Or even better put the debug var to blackbox. I usually use the attitude[] array in blackbox to view my debug data.

borisbstyle, 2015-12-18:
You can use fast_pwm feature within betaflight and set the motor update to
4khz

set fast_pwm = ON (or 1)
set motor_pwm_rate = 4000

lkaino, 2015-12-18:
Bit off-topic here but I would like to ask you Boris how do you debug your stuff while developing? Do you have a HW debugger or use blackbox, are there other possibilities like printing to serial?

borisbstyle, 2015-12-18:
Depending of what I have to debug I usually use MSP debug or sometimes I
hack blackbox to show me things I want to see.

anyway try fast_pwm you should get perfectly nice timing there up to 4khz.

universam1, 2015-12-18:
Sounds good. How to shift starting point?

BTW to make BF in a "balancer mode" without the unnecessary stuff, how to
do without messing all around the code?

universam1, 2015-12-20:
@borisbstyle Are you sure about the fast_pwm I could use for?
Since this messes up system-wide access to the ESC which I still need for running motors...

borisbstyle, 2015-12-20:
How do you mean system wide access?

You need to have fast pwm enabled together with oneshot btw.

Yes fast pwm works. It sets refresh rate outside the loop.

universam1, 2015-12-20:
I mean that I would need the fast pwm only for the LED port, but the ESC/Motor ports are not to be changed. I want a solution that fits other people too. Or am I misunderstanding you?

Looks like some timer >4 could be used for an IRQ routine...

borisbstyle, 2015-12-28:
The latest code has 2khz support.

borisbstyle, 2016-01-20:
Closing this as I didnt hear anything more about it

closed 2016-01-20 by borisbstyle