betaflight/betaflight #67: Thoughts on PID integration and runaway behaviour

martinbudden, 2015-12-15:
It occurred to me that the cause for runaway behaviour could be that the PID itegrator uses simple Euler integration.

Sure enough we have:

    // -----calculate I component.
    errorGyroIf[axis] = constrainf(errorGyroIf[axis] + RateError * dT * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);

or, taking its essence:

    I += error * dT

a simple Euler integration.

Euler integration is a simple first order method and can be numerically unstable:

https://en.wikipedia.org/wiki/Euler_method#Numerical_stability

"The Euler method can also be numerically unstable, especially for stiff equations, meaning that the numerical solution grows very large for equations where the exact solution does not."

I posit that replacing the Euler method with something higher order (for example Runge-Kutta: https://en.wikipedia.org/wiki/Runge%E2%80%93Kutta_method) would increase stability and reduce runaway behaviour. Runge-Kutta can be unstable, but it is much less unstable than Euler.

RK4 is the generally used method. It is fourth order and the accumulated total error is O(h^4), i.e. for betaflight O(dT^4). In contrast Euler integration is first order.

The disadvantages of RK4 is that it requires more computation (although not that much more) and requires the past three RateErrors (for each axis) values to be stored.

An equivalent way of looking at this is that currently the numerical itegration is done using the "rectangle method" https://en.wikipedia.org/wiki/Rectangle_method whereas using RK4 would be doing the integration using Simpson's rule https://en.wikipedia.org/wiki/Simpson's_rule.

Alternatively, if RK4 is too complicated, a second order integration method could be used - this would still produce less runaway than Euler.

martinbudden, 2015-12-15:
I've just looked at the code a bit more and realised that the `lastError` is already stored since it is needed for the D-term calculation. This means we can easily replace the [rectangle method](https://en.wikipedia.org/wiki/Rectangle_method) integration with a [trapezoidal method](https://en.wikipedia.org/wiki/Trapezoidal_rule) integration. Just replace:

        errorGyroIf[axis] = constrainf(errorGyroIf[axis] + RateError * dT * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);

by:

        errorGyroIf[axis] = constrainf(errorGyroIf[axis] + 0.5*(lastError[axis] + RateError) * dT * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);

[Note we are just replacing `RateError` with `0.5*(lastError[axis] + RateError)`, that is the average of it and the previous error.]

This makes the integration second order and should reduce runaway behaviour. It would then be possible to experiment with reducing the constraints.

[Note: to do RK4 we'd have to store `errorBeforeLastError[axis]`]

martinbudden, 2015-12-15:
Further comment. I've just seen that `dT` is not constant. So you'd need to store `lastdT` and use:

    (lastError[axis]*lastdT + RateError*dT)/(lastdT + dT)  // weighted average of errors
instead of

    0.5*(lastError[axis] + RateError)
so the expression becomes:

    errorGyroIf[axis] = constrainf(errorGyroIf[axis] + ((lastError[axis]*lastdT + RateError*dT)/(lastdT + dT)) * dT * pidProfile->I_f[axis] * 10, -250.0f, 250.0f);

borisbstyle, 2015-12-15:
dT is constant on betaflight :)
I made it to be constant as betaflight operates exactly at 1khz.

Can you perhaps explain the "runaway" behaviour better. You mean the Iterm
will fluctuate or quickly jump from its offset?

If possible you can also propose a PR and prefferably test it one way vs
another and see if you can see any significant improvement?

martinbudden, 2015-12-15:
By "runaway" behaviour I mean accumulation of error over time. I'm not sure of you maths background so not sure what level to pitch the explanation. My apologies if this is too basic.

In the PID controller I is the integral of P. It is approximated by accumulating `error*dt`. Although each error may be small they can accumulate:

[image]

Blue is the true value of the integral, red is the approximation.

The term "windup" is used in the code, and in the release notes you talk of "building up of Iterm". The code uses the `constrainf` function to limit this. This is the thing that I am calling "runaway" (actually it's not my term, but I have seen other people use it in discussions).

The `constrainf` function limits the windup, but does not address its root cause, which I believe to be the fact that a first order integration method is used.

Unfortunately I am not going to have much flying time over Christmas, so I doubt if I'll have a chance to test this (I'm also only a beginner pilot). I'm not sure if the improvement will be obviously visible in flight, but it should be visible in the `axisPID_I` terms in the blackbox.

I can submit a PR if you like, but realise it is a bit of a cop out to submit a PR without testing it myself.

borisbstyle, 2015-12-15:
I understand what I does. That code is not mine actually. This is actually
original baseflight->cleanflight->betaflight part.

In my case the windup occured only on the ground as the pid loop is active
there with new feature "Air mode". That windup is perfectly normal as the
copter is not flying.

In flight windup did as far as I know never occure so I think the first
order behaviour is good for now unless it can be proven that it isnt.

You can suggest your PR anyway if you want and I can test. All new ideas
what could potentially improve things are welcome.

martinbudden, 2015-12-15:
I've already forked cleanflight, and it seems github does not support forks of forks. So here's a patch instead. I hope that is ok.

[pidintegration.patch.txt](https://github.com/borisbstyle/betaflight/files/63301/pidintegration.patch.txt)

I'll be interested to see if you think it makes an improvement.

Thinking about this a bit more, I think this might more useful for the PID in altitude hold, where people do seem to have runaway problems during flight. And that is something I can test myself, and, if it works, results should be visible. So a little project for me in the new year.

borisbstyle, 2015-12-16:
@martinbudden 
Thanks. Betaflight has 2 PID controllers, but I will add to PID1 for testing. This is pretty much just averaging over 2 errors

borisbstyle, 2015-12-16:
You do can fork betaflight separately btw

martinbudden, 2015-12-16:
I just noticed that `lastError` is not initialised in `pidLuxFloat`

`static float lastError[3] = { 0.0f, 0.0f, 0.0f };`

is required in the declaration. (It is initialised in `pidRewrite` though.)

lkaino, 2015-12-16:
According to The Standard, objects with static storage duration are automatically initialized to zero 
unless initialized explicitly.

Like your ideas btw, would be nice to see what kind of difference would the RK have. The I behaves quite well, but it could do it even better:).

borisbstyle, 2015-12-16:
I added this code to both PID1 and PID2. I tested it and

borisbstyle, 2015-12-16:
I added this code to PID1 and PID2. I did test it. I didnt notice any noticaeble difference as I expected, but the theory behind this proposal sounds so reasonable to me that I just added it and will be released in 2.1.6
https://github.com/borisbstyle/betaflight/commit/9a83df04f52fdd51bfb9970f31e26cf3679f08fb

closed 2015-12-16 by borisbstyle