betaflight/betaflight #69: Yaw issue

beb38138, 2015-12-18:
When i yaw quickly left or right, the quad falls to the ground. If im gentle on the yaw, it wont do it. Im using the naze32 rev5, and tested with 3 different boards, all with same issue. I set yaw rate to .30

borisbstyle, 2015-12-18:
This is probably hardware related. Check your ESC's for example.

closed 2015-12-18 by borisbstyle

SoulPictures, 2015-12-18:
Check If you get sync errors at fast changes, it is more likely that your esc and motors cause this error than that it is caused by betaflight

universam1, 2015-12-18:
can confirm this behaviour, anything above a yaw rate of .12 cant hold altitude, the higher the worse. 
.30 would be fall to ground.

borisbstyle, 2015-12-18:
Yaw is the most agressive axis for the motors. This will reveal hardware issues.

Blackbox it and you will see

universam1, 2015-12-18:
could you explain that a little bit more?
The ESCs are working nicely over Oneshot, no issues known. What do you expect to be?
On high YAW rate the two motors kind of stop, so probably the others are not strong enough to keep altitude at half thrust available. Does this make sense?

beb38138, 2015-12-18:
I understand. It very well could be loosing sync. I flashed a stable version to the naze from the drop down menu in cleanflight and the issue is gone. I may try updating my bl20a escs to a newer version of blheli (currently on 14.1) and see if that helps. Also I will drop the yaw rate down from .30, thanks for your help!

jtp755, 2015-12-18:
@universam1, @beb38138,

Can you guys add a bit more detail to how you are creating the issue?

jtp755, 2015-12-18:
I just tested with a Naze32 Acro rev5 and 2.1.6-RC1 and had no issues with yaw. I tested in acro with and without air mode as well as horizon mode with and without air mode.

beb38138, 2015-12-18:
The issue has been resolved. I updated to blheli 14.3.01 and changed yaw rate to .20. Seems to be doing fine now. Would you still like more info? If so what would you like to see?

jtp755, 2015-12-18:
Im running BLHeli 14.1 and rates at .35.

I was just going to help with testing and if I were you I would be curious why it was not working =P

Did you re-calibrate ESCs after upgrading to 14.3? That may have got things back in order. 

I would suggest bumping the yaw rate back up to where it was and see if the issue is still there.

beb38138, 2015-12-18:
I turned yaw rate to .35 and still no issues. Maybe it had something to do with my super cheap bl20a escs. Definitely glad I can run beta again now. ;)

jtp755, 2015-12-18:
Great! Just for reference I am running Spider Opto 12As (Atmel chip and SimonK bootloader)

Now you can get back to flying...

universam1, 2015-12-18:
Thanks for help, BLHeli seams to be the solution. So was a hardware issue.