betaflight/betaflight #73: Settings are lost after turning on AIR MODE mode

topchiyev, 2015-12-23:
I've read about new AIR MODE and decided to turn it on on my ZMR250. I use 3s and 4s batteries on the same copter, so I have a bit different PIDs, RC Rates, Pitch/Roll/Yaw rates, TPA, etc. And I differentiate them storing in Pid profiles 1 and 3.

How to meet the issue:
1. Backup settings
2. Turn off MOTOR_STOP
3. SAVE & REBOOT
4. Set Pid profile to 3
5. Save
6. Turn on ARM, set active zone to 1300 - 2100
5. Turn on AIR MODE, set active zone to 1300 - 2100
6. Save
7. Set Pid profile to 1. Writes in log "Switched to profile 1", but 3 is selected in the combobox.
8. Close the Cleanflight configurator, disconnect USB (battery not connected)
9. Reconnect USB, open Cleanflight configurator, hit connect
10. All modes are lost, all Adjustments are lost

I thought may be I should reset the board and manually set every thing up without restoring from backup.

And I tested it. And as long as I do not activate the AIR MODE everything works OK, saves, reads, etc. And when I activate the AIR MODE mode everything happens again.

My gear:
Naze32 Acro black,
Cleanflight configurator 1.1.0
Betaflight 2.1.6

Thanks.

borisbstyle, 2015-12-23:
Can you check this affter resetting all the settings?

COnfigure everything without air mode. See if it works with 2 profiles and send me the dump of both profiles before you configure air mode.

topchiyev, 2015-12-24:
OK. Here are the dumps.
[cleanflight-profile-0.txt](https://github.com/borisbstyle/betaflight/files/71919/cleanflight-profile-0.txt)
[cleanflight-profile-2.txt](https://github.com/borisbstyle/betaflight/files/71920/cleanflight-profile-2.txt)

topchiyev, 2015-12-24:
But here I've turned on the MOTOR_STOP.

borisbstyle, 2016-01-04:
is this still an issue btw?

borisbstyle, 2016-01-13:
Didnt hear anything about this. Please open a new issue if the problem persists in the latest firmware

closed 2016-01-13 by borisbstyle