betaflight/betaflight #74: Main PIDC (rewrite) updated default values a bit high for 180 quads?

varoudis, 2015-12-23:
You have do a couple of updates lately to the default PID values, based on pilot comments, but I think they are not good for small 180 quads.

I get some oscillation after some stick moment or when the quad tries to re-level it self on horizon mode.
On acre you can see it with a bit of wind. The quad if 'over'-trying to keep the given tilt/trajectory.

If you think its something else or if you need some more specify feedback let me know.

At the moment Im flying 2.2.0 compiled from source.

borisbstyle, 2015-12-23:
enable rc_smoothing

set rc_smoothing = ON

varoudis, 2015-12-23:
ok will test tomorrow.

Not sure if its 'input' related though as for maintaining trajectory the oscillation happens without input if I remember correctly. Not sure though! I have to double check.

Thanks

borisbstyle, 2015-12-23:
Yes its because rc_smoothing is not active.

I added to the comment that level/horizon fliers should better enable it.

borisbstyle, 2015-12-28:
I will close this one and enable rc_smoothing by default in the next release.

closed 2015-12-28 by borisbstyle