betaflight/betaflight #75: CC3D Tricopter Servo Mapping

catchthewind, 2015-12-26:
Could you please update CC3D tricopter servo mapping in Betaflight. It was broken or never implemented into Cleanflight till 1.10. Github link https://github.com/cleanflight/cleanflight/issues/586
Apparently the newest CF 1.11 has fixed it for the first time to use on CC3D but this is also first version of Cleanflight to remove BIN and openpilot bootloader support

Could you please fix this mapping and put it into PIN 4 to use together with buzzer on PIN 6.
New Cleanflight 1.11 should have it ready to use but I can not get it to work in Betaflight 2.1.6

Thanks in advance.

borisbstyle, 2015-12-28:
I think it should be fixed in here

https://github.com/borisbstyle/betaflight/commit/7a3acbabef7b4ebf33e6ac68121012b36fdb3659

closed 2015-12-28 by borisbstyle

catchthewind, 2015-12-28:
Thank you!!!!
I was going to get the details for you but was waiting for confirmation.
I think this  https://github.com/cleanflight/cleanflight/pull/1042/files  should do the trick.
I have found it here https://github.com/cleanflight/cleanflight/issues/586

catchthewind, 2015-12-28:
@borisbstyle 
 Here is exact fix given by a6j https://github.com/cleanflight/cleanflight/commit/165adfa755053e68224b9efa664fa6052372a3cf
Is there any eta date for 2.2.0 ??
Thank you!!

borisbstyle, 2015-12-31:
@catchthewind 
Can you try RC2 and verify if it works?

catchthewind, 2015-12-31:
@borisbstyle 

I have just tried it and it flies fine on a tricopter!!!

I had to lower YAW P to 5 (normal for tricopters)
I had to lower LEVEL angle & horizon to 3 and its still a little bit high. I will try 2,5 or 2.
On acro it is great and stable. Very fast and predictable YAW after changing digital servo to 400

set servo_pwm_rate = 400 (could be 333 depending od the servo used)
set servo_pwm_rate = 50  (default) is also ok but used mainly for analog and slower servos

I have found on 3-4 different quads and a tricopter that on Controller Multiwii (rewrite) you need to lower LEVEL to 2 or 3.  Default level of 5 or 10 (before) is not flyable on rewrite and creates shaker  ;)
You might consider lowering DEFAULT Level values on Controller Multiwii (rewrite) for newcomers.

Thanks a lot for ALL your great work.  I have just made beer donation :)