betaflight/betaflight #77: set telemetry_inversion in CLI only accepts on/off instead of 0/1 as acceptable values

carsonblack, 2015-12-28:
Entering set telemetry_inversion = 1 in the CLI returns "Invalid value" 
Entering set telemetry_inversion = on returns "telemetry_inversion set to ON" and behavior is as expected. Only reporting because documentation advises using the integer value. Not sure if it's a problem with the code or the docs. 

Thanks for all your work on this great software!!!

borisbstyle, 2015-12-28:
Yes ON/OFF is the new CLI style.

Betaflight doesnt have a full own documentation. It uses cleanflight documentation, which needs an update

closed 2015-12-28 by borisbstyle

carsonblack, 2015-12-28:
So the cleanflight docs should be updated? And you will pull them in (eventually) to this project? or do the docs need to be updated in both places separately?