betaflight/betaflight #81: overloading ACC/GYRO causing reboot sequence?

varoudis, 2015-12-29:
Hi,
I changed my naze32 to a SPF3 and now if a shake the quad with a bit of force I get a FC reboot!!!!
I don't thing its a connection problem, this is why Im posting this as a possible software thing, because:

If the quad is _NOT_ ARMed (via channel switch ) it will not reboot. All OK if a shake it or slam it.
If the quad is ARMed, with or without motors running (with ARM on, I need to get the throttle LOW to start motors), I will get a reboot.

The only thing connected to SPF3 is an X4R SBUS RX. Nothing else on it, no sensors no other cables attached. (only 4 ESCs Oneshot  )
Running 2.1.6 Betaflight. Default settings apart from SerialRX, ARM+HORIZON AUX and custom MIXER.

Any ideas or debug steps that might help??

borisbstyle, 2015-12-29:
This sounds like a typical hardware issue. Armed copter draws more current.
Maybe your bec is failing. Its not the software.

I have 2 identical setups with this board and no issue with it.

Try also with cleanflight :)

varoudis, 2015-12-29:
"Armed copter draws more current"

What applies to ARMed but NO motors running? My software guy brain might be missing something as it seems like the same thing running. (Might be trivial but I'm on holidays with no good multimeter to check).
Thanks

PS: Ill close it but if you can comment on the above.

closed 2015-12-29 by varoudis

borisbstyle, 2015-12-29:
I have no idea, but your issue sounds like very mechanical rather than software :)