betaflight/betaflight #85: Name of pidMultiWiiRewrite or pidRewrite

martinbudden, 2015-12-29:
This is not really a very good name. It reflects the history of how the function was developed rather than what it does. And it's been rewritten so much that it bears very little resemblence to `pidMultiWii23.`

The function is, in essence, an integer version of `pidLuxFloat` (the code is very similar). So I suggest it is renamed to `pidLuxInt`.

borisbstyle, 2015-12-29:
It is not the same and especially not to the users as the tuning numbers are completaly different.
pidRewrite is kind of famous under the public so I would definately not change that.
In your suggestion it would sound like it was inspired from luxfloat, while it is totally the other way around.

martinbudden, 2015-12-29:
Fair enough. I had to ask though.

closed 2015-12-29 by martinbudden