betaflight/betaflight #87: P Inflight adjustment only increases very slow

LeChuck71, 2015-12-30:
Like the title says the inflight adjustment of "P" increases very slowly i.e. 0.1 in 7 seconds. With I and D we get the rates like in cf (like twice a second).

Also the docs say the the value changes at least once if you trigger the adjustment switch. This does not work on the "P" setting - only I and P adjust once you trigger it.

I attached a dump of my system, maybe it can help. Adjustment "selector" is a poti with "low" P setting, "middle" I setting and "high" D setting. Up/Down via 3Pos switch

[BetaFlightCC3D 2.1.6.txt](https://github.com/borisbstyle/betaflight/files/74673/BetaFlightCC3D.2.1.6.txt)

fedorcomander, 2015-12-30:
it increases 0.01 ;) so ten times will make you 0.1 ;)

LeChuck71, 2015-12-30:
I am quite sure that the "P" value increased 0.1 in cleanflight, because I could see a change if I triggered the switch just once.
I value increases 0.001 an D in 1 steps.
Can I change the increments somewhere?

SoulPictures, 2016-01-02:
Similar behavior here: P increases 0.01 ( which seems to be nice to fine tune, but GUI does not reflect such small changes yet) 

Yaw Rates jumps 0.12 per Beep, which seems a bit overkill for me (Pitch and Roll Rate are not affected, standard 0.01 change).

Setup: Dodo (SP3 hex, RC2.2.0.RC2), X4R (SBUS)

borisbstyle, 2016-01-02:
This is a general cleanflight issue. It should be opened there.

closed 2016-01-02 by borisbstyle