betaflight/betaflight #88: 2.2.0 RC1 on CC3D acting WEIRD, motors Spinning FULL Speed after arming :(

catchthewind, 2015-12-30:
I am using BF on my 2 flip32 board in 2 quadcopters on few last versions upto 2.1.6 without problems

I wanted to reuse my CC3D board on a tricopter and I have asked you to fix that first https://github.com/borisbstyle/betaflight/issues/75

I have never used Betaflight on this CC3D but board is OK and tested yesterday on new CF 1.11
Today I wanted to try newest BF 2.2.0 RC1 with fixed tricopter mapping.

After uploading hex FULL CHIP ERASE option everything looks OK till Arming.
Settings WERE reset to default few times 
After Arming all engines go full power and do not react on throttle, pitch, roll, yaw or moving the copter.
When looking on motor TAB 1st engine shows 1150 but goes full power, 2nd and 3rd shows 1850
Setup tab shows copter movement ok.
I have reflashed CC3D twice just in case with newest HEX
Receiver Tab is ok and all channels are working OK
Servo is active when I change RATE on SERVO tab but doesnt move when I move tricopter or change yaw  (should be moving automatically an on yaw input)

I have tried both tricopter and quadx settings.

It looks like a problem not connected to tricopter. 
Maybe broken hex or broken CC3D version of the firmware.
I am using DSMX sattelite which was ok with CF 1.11 and many others.

borisbstyle, 2015-12-30:
Yes this is reported by many. It seems there is an issue with CC3D due to those special changes

waslater, 2015-12-30:
Not just on CC3D.  I see this behavior on two Naze32 rev5 ships. see here: http://www.rcgroups.com/forums/showpost.php?p=33582182&postcount=12794

catchthewind, 2015-12-30:
I have tested BF 2.1.6 on this CC3D right now and it is working OK without SERVO but MOTORS are OK. The problem is in new 2.2.0 RC1 somewhere.

Im not sure if its new tricopter mapping to give this problem or something else.

I just wanted to confirm that I have tried newest CF 1.11.0 and tricopter flies fine on CC3D. I dont know how hard it is from you to use the code but just informing that it works fine there.

borisbstyle, 2015-12-30:
@waslater
naze32 just works.

borisbstyle, 2015-12-30:
@catchthewind
That code is one of the suspects.

I will provide you some binaries to try to see if we can find out what
change caused it.

waslater, 2015-12-30:
@borisbstyle both of mine certainly don't. Especially on rewrite. I'll post a video on RC groups to document further.  I haven't tested my dodo, but I'll do that too.

borisbstyle, 2015-12-30:
if it is only rewrite than it is for 100% different issue than this.
Try lowering D on rewrite perhaps.

waslater, 2015-12-30:
Lux is has strange blips in it too, but at least doesn't go full throttle.  I will lower D to 0 and test again on rewrite.

waslater, 2015-12-30:
That did it on PIDC1 and naze32, if I zero out yaw on all axes, then no jump to full throttle.  If I set all D values at 5, problem comes back. Want me to open a new issue?

borisbstyle, 2015-12-30:
Yes maybe good idea to separate it but collect as much as possible info.

type tasks in cli and type status. See what it says there.

Also report your gear etc.

Your issue seems only 2khz related right?

waslater, 2015-12-30:
No it happens on either gyro lpf OFF, or 188hz

catchthewind, 2015-12-30:
I am ready to test some hex for you.
I have done some more testing and tried to disable failsafe, enable and disable motorstop and oneshot. It doesnt change anything. 

Every time I arm engines 1,3,4 goes 1850 and 2 goes 1150 on quad
On tricopter motor 1 goes 1150 and motors 2,3 goes 1850, servo locked

@borisbstyle 

I have changed all D to 0 and it fixes EVERYTHING on Multiwii (Rewrite)
Motors works like they should be. Servo reacts on yaw and copter movements!!
D is the problem

borisbstyle, 2015-12-30:
Did you try enabling dterm_cut_hz?

waslater, 2015-12-30:
This is very similar to my issue... if I zero D-term the throttle jump disappears. I will try enabling dterm_cut_hz

waslater, 2015-12-30:
I just set dterm_cut_hz = 40  and kept my usual numbers on D-term and that fixed the throttle jump...

catchthewind, 2015-12-30:
I just set dterm_cut_hz = 1 and that fixed the problem. What value should I put here?

henn1001, 2015-12-30:
Confirmed here http://www.rcgroups.com/forums/showpost.php?p=33584020&postcount=12833
@borisbstyle  this still makes me think that there could be something wrong in this merge https://github.com/borisbstyle/betaflight/commit/66cd5e86b6e5eff92bde2bd1007f4bb4e627ee32

borisbstyle, 2015-12-30:
@henn1001 
I dont think its that one. That is a simple change, but there were many others around that code. I will check that part again.

borisbstyle, 2015-12-31:
Fixed in RC2
https://github.com/borisbstyle/betaflight/releases/tag/v2.2.0-RC2

closed 2015-12-31 by borisbstyle

catchthewind, 2015-12-31:
Tested on a tricopter &CC3D and works OK!!

catchthewind, 2015-12-31:
I have also tested enabling beeper on PIN6 with 2.2.0 RC2

Tricopter works fine, beeper beeps fine alarms and notification. Servo works fine :)

Only problem is that beeper is giving constant silent noise even without the alarm.
I am using simple active beeper 3-5V connected to middle and upper pin of output 6.
Do I need any special beeper for CC3D?? Link please.

All the best and thank you for your work and help.

henn1001, 2015-12-31:
Uhm did you just connect the beeper to output 6? You should use a transistor for that because the output of the stm32 chip doesnt deliver enough current.. 
See here for example https://github.com/cleanflight/cleanflight/blob/master/docs/Wiring/CC3D%20-%20buzzer%20circuit.pdf
Pin38=MotorP6  - Any PNP Transistor should do.

catchthewind, 2015-12-31:
Ok, thanks :)

mhusiak, 2017-09-22:
(Excuse my etiquette, it's my first time using github)

I'm having this same problem on my CC3D Atom on 3.0.1. It didn't work the first time I flashed, worked after trying different releases and settling on 3.0.1, and then the weird full throttle thing started again after I adjusted my rates (unrelated?).

I thought that what fixed the issue was adjusting the gyro/pid frequency to 1kHz/0.5kHz. But that isn't helping now.

Has this been identified and fixed?

jflyper, 2017-09-23:
@mhusiak 

- This issue has been raised and fixed almost TWO years ago, and it was related with v2.2.0.
- Please redirect your questions and support requests to some appropriate forums on the net, as Betaflight github is not a place for support.
- FYI, please note that F1 support in Betaflight is coming to EOL.
https://github.com/betaflight/betaflight/wiki