betaflight/betaflight #89: 2.2.0 RC1 -- Throttle jump on arm

waslater, 2015-12-30:
This was initially documented here: http://www.rcgroups.com/forums/showpost.php?p=33582182&postcount=12794

Using a Naze32 rev5, with D4R-II RX, and mostly default values, the motors jump to what sounds like full throttle when armed. Airmode has generally the same behavior. 

The dump and BB log are at link above. Tasks list is here:

Entering CLI Mode, type 'exit' to return, or 'help'

# tasks
Task list:
0 - SYSTEM, max = 11 us, avg = 0 us, total = 59 ms
1 - GYRO/PID, max = 574 us, avg = 259 us, total = 477755 ms
3 - SERIAL, max = 59567 us, avg = 2 us, total = 825 ms
4 - BEEPER, max = 8 us, avg = 0 us, total = 135 ms
5 - BATTERY, max = 40066 us, avg = 1 us, total = 261 ms
6 - RX, max = 113 us, avg = 53 us, total = 3164 ms

# status
System Uptime: 892 seconds, Voltage: 8 * 0.1V (0S battery - NOT PRESENT), System load: 0.06
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 494, I2C Errors: 0, config size: 2020

# 

Here is a video of the issue: https://www.youtube.com/watch?v=txcji0HJCGI

waslater, 2015-12-30:
This is the issue using these settings and Lux:
https://www.youtube.com/watch?v=13BBQIXDdow

borisbstyle, 2015-12-30:
Oh wait....thats not an issue.

Use:
set rc_smoothing = On to remove it

thats derivative kick. Phenomenon where D spikes due to control input.
However it doesnt effecr the flight itself usually

waslater, 2015-12-30:
Ah that makes sense, any idea about the problem with PIDC1?

borisbstyle, 2015-12-30:
maybe excess D?

Did you try dterm_cut_hz for a change on pid1?

waslater, 2015-12-30:
I just set dterm_cut_hz = 40 and the throttle jump went away

borisbstyle, 2015-12-30:
@waslater 
Well we are defining new defaults :dancer: 

Me and few others had better experience in flight without it. You  perhaps could lower D without dterm_cut_hz. I use value of 15-25 with success

waslater, 2015-12-30:
@borisbstyle Well I actually tried D values of 1 on each axis with dterm_cut_hz = 0 (only on PIDC1).  even that caused the throttle jump. 

So there are NO D values that will work when using rewrite + dterm_cut_hz = 0 + Naze.

Keep in mind this is only the naze target, I don't know why it would act differently than my dodo, which does not have this issue.

borisbstyle, 2015-12-30:
that is indeed weird.

the issue is on 1khz mode too?

waslater, 2015-12-30:
Yes, its the same with set gyro_lpf = 188HZ

waslater, 2015-12-30:
I also tried setting all D values to 1, and playing with the dterm_cut_hz value.  Any value other than 0 seems to work.  Went all the way to dterm_cut_hz = 200.

borisbstyle, 2015-12-30:
ok......that is a very usefull info for me!

waslater, 2015-12-30:
glad to help!

digitalentity, 2015-12-30:
@borisbstyle 
Here https://github.com/borisbstyle/betaflight/blob/v2.2.0-RC1/src/main/flight/pid.c#L203-L207

`deltaSum` is not  reset to zero prior to adding new values to it here https://github.com/borisbstyle/betaflight/blob/v2.2.0-RC1/src/main/flight/pid.c#L206

Reset it to zero beforehand and you are all good.

waslater, 2015-12-30:
Any idea why this problem only manifests on naze and not on dodo running spf3 hex?

borisbstyle, 2015-12-30:
@waslater 
That's indeed the interresting part. My F3 boards all work fine

digitalentity, 2015-12-30:
That's a mere coincidence. By a lucky chance deltaSum is allocated stack
with values close to zero on an F3 targets.

borisbstyle, 2015-12-30:
Yes! just realized that 10min ago!

borisbstyle, 2015-12-30:
I reproduced it and fixed the bug. Going to update to RC2

henn1001, 2015-12-30:
yay for progress D:

borisbstyle, 2015-12-30:
Omg going to kill myself for this one...

https://github.com/borisbstyle/betaflight/commit/560bd745f03cb4ea2e082942064162952f849050

borisbstyle, 2015-12-30:
I basically wanted to make the ugly averaging neater and continued simplifying that code and than this happened!
My yesterdays testing was my F3 boards, which was all successfull

digitalentity, 2015-12-30:
Please don't kill yourself! Who else would design new cool features and
lead the greatest Cleanflight beta testers community ever!

borisbstyle, 2015-12-30:
@digitalentity 
Yeah this crap happens, but damn watched that part of the code like 20 times today and didnt see it :D

borisbstyle, 2015-12-30:
Fix is here!
https://github.com/borisbstyle/betaflight/releases/tag/v2.2.0-RC2

closed 2015-12-30 by borisbstyle

digitalentity, 2015-12-30:
Fresh look! Woke up today, seen this issue, gave it a thought over a
morning coffee and posted here :)

waslater, 2015-12-30:
Thanks for the fix!

borisbstyle, 2015-12-31:
@waslater 
So to explain the issue to you.
Well basically deltaSum was not being reset so the roll D was rolling into pitch and yaw etc. deltaSum was not only axis based, but mixed through all axis :D. Pretty nasty.

waslater, 2015-12-31:
Haha that sounds quite nasty.  Like D spilling out everywhere....  Anyway thanks for the fix, and I'm glad I could assist!