betaflight/betaflight #92: betaflight 2.2.0 rc2 naze32 issue

raymonkroeze, 2015-12-31:
If i enable 2khz on my naze32, everything starts to act strange. The first thing i notice was, it takes a longtime before my esc's starts up. and after running the motors a while, the motors en esc's will be much warmer the usual. 
I disabled the acc, baro and mag.
with normal 1khz it flies amazing and everything acts the way it should.

hope this is usefull.

borisbstyle, 2015-12-31:
It sounds like your esc doesnt support 2khz properly. Not a firmware issue.

What esc firmware?

raymonkroeze, 2015-12-31:
im using little bee 20A esc's with blheli 14.1

borisbstyle, 2015-12-31:
upgrade to blheli 14.2 and try again

borisbstyle, 2015-12-31:
or 14.3....thats fine too

raymonkroeze, 2015-12-31:
I will try that tomorrow! thank you for the fast reply!

raymonkroeze, 2016-01-01:
no im using 14.3 firmware, but im still having the issue. im using a naze32 acro. not full. don't know if that causes any issues.

borisbstyle, 2016-01-01:
Can you post your dump and tasks output. Type in tasks in cli and also type
status and post the output of that here.

raymonkroeze, 2016-01-02:
Entering CLI Mode, type 'exit' to return, or 'help'

# tasks
Task list:
0 - SYSTEM, max = 9 us, avg = 0 us, total = 7 ms
1 - GYRO/PID, max = 477 us, avg = 333 us, total = 38320 ms
3 - SERIAL, max = 67 us, avg = 2 us, total = 37 ms
4 - BEEPER, max = 14 us, avg = 0 us, total = 64 ms
5 - BATTERY, max = 118 us, avg = 1 us, total = 21 ms
6 - RX, max = 95 us, avg = 64 us, total = 485 ms

#

raymonkroeze, 2016-01-02:
# status
System Uptime: 139 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.02
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 993, I2C Errors: 0, config size: 2020

raymonkroeze, 2016-01-02:
Im using the luxfloat controller. Don't know if that is causing any problems. I already ordered a spracing f3 FC. So i expect better results with that FC.

borisbstyle, 2016-01-02:
are you using rev6?

raymonkroeze, 2016-01-02:
This is a rev5 board from mrp.

borisbstyle, 2016-01-02:
you are the only one with this issue.

What exactly was the problem again?

borisbstyle, 2016-01-02:
did you tey pidrewrite?

raymonkroeze, 2016-01-02:
I already reflashed the FC same issue. Only problem is that it's taking a longtime before are the esc's are powered up. Also the esc's and motors are getting warmer than usual. For this reason i don't dare to fly it. But al the motors seems to react normal to stick inputs..

borisbstyle, 2016-01-02:
try rewrite

borisbstyle, 2016-01-02:
closing this one as this is the same issue
https://github.com/borisbstyle/betaflight/issues/94

closed 2016-01-02 by borisbstyle