betaflight/betaflight #94: Naze32 rev5 wont arm after disabling accelerometer

carebear482, 2016-01-01:
When I enable accelerometer via "set acc_hardware=0" I can arm again.

Here is my dump:
# version
# BetaFlight/NAZE 2.2.0 Dec 31 2015 / 00:50:34 (560bd74)
# dump master

# mixer
mixer CUSTOM
mmix reset
mmix 0 1.000 -1.000 0.707 -1.000
mmix 1 1.000 -1.000 -0.707 1.000
mmix 2 1.000 1.000 0.707 1.000
mmix 3 1.000 1.000 -0.707 -1.000
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature ONESHOT125

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set emf_avoidance = OFF
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 8
set rssi_scale = 30
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set rc_smoothing = OFF
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set enable_fast_pwm = OFF
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set serialrx_provider = SBUS
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = -90
set max_angle_inclination = 700
set gyro_lpf = OFF
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set yaw_control_direction = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set rx_min_usec = 885
set rx_max_usec = 2115
set acc_hardware = 1
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set beeper_off_flags = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 0 0 1300 2100
aux 1 12 1 1300 1700
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = NORMAL
set acc_cut_hz = 15
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_controller = LUX
set p_pitch = 40
set i_pitch = 30
set d_pitch = 20
set p_roll = 40
set i_roll = 30
set d_roll = 20
set p_yaw = 100
set i_yaw = 50
set d_yaw = 8
set p_pitchf =  1.500
set i_pitchf =  0.300
set d_pitchf =  0.010
set p_rollf =  1.500
set i_rollf =  0.300
set d_rollf =  0.010
set p_yawf =  4.000
set i_yawf =  0.400
set d_yawf =  0.010
set level_horizon =  6.000
set level_angle =  6.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set gyro_soft_lpf = ON
set dterm_cut_hz = 0
set yaw_pterm_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 120
set rc_expo = 50
set rc_yaw_expo = 30
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 50
set tpa_rate = 0
set tpa_breakpoint = 1500

#

borisbstyle, 2016-01-01:
disabling accelerometer puts it in 1khz mode of course.

Can you type the other 2 commands I gave you

borisbstyle, 2016-01-01:
try also these 2 things:

try arming with pidrewrite instead of luxfloat. See if that works.

try arming through stick when disabling arming on aux

carebear482, 2016-01-01:
Yep, it seems to be related with lux pidcontroller. When switching to rewrite, I can arm again.

borisbstyle, 2016-01-01:
ok your board must have some kind of performance issue.

can you swap to luxfloat and type 2 things:
status

than type
tasks

Also is this a real naze32 rev5?

carebear482, 2016-01-01:
Yes, it is naze32 rev5, with gyro/acc only. 
Here is status and tasks:
# status
System Uptime: 141 seconds, Voltage: 152 * 0.1V (4S battery - OK), System load: 3.63
CPU Clock=72MHz, GYRO=MPU6050
Cycle Time: 491, I2C Errors: 0, config size: 2020

# tasks
Task list:
0 - SYSTEM, max = 16 us, avg = 0 us, total = 10 ms
1 - GYRO/PID, max = 580 us, avg = 405 us, total = 83979 ms
3 - SERIAL, max = 63507 us, avg = 84 us, total = 1002 ms
4 - BEEPER, max = 16 us, avg = 0 us, total = 30 ms
5 - BATTERY, max = 40070 us, avg = 1 us, total = 72 ms
6 - RX, max = 154 us, avg = 92 us, total = 1293 ms
12 - TELEMETRY, max = 16 us, avg = 0 us, total = 183 ms

borisbstyle, 2016-01-01:
Still your naze has a very bad performance in luxfloat. Somehow your board
seems slower than my tested naze boards here.
405us average + 92us rx is already not able to made within 500us
Your rx tasks gets discarded and you have no control

I recommend to use PID1 on 2khz in general and especially on your board as
it is faster on F1 boards.
Luxfloat uses floating point math and Naze and other F1 boards are really
poor on that.

You can try one more thing and that is overclocking your board a bit with.
set emf_avoidance = on

But you can see by yourself that you are on the edge of losing your rx task.

carebear482, 2016-01-01:
Thank you for explaining. I think it is time to move to f3/f4 board, but I'll try this out in PID1.
Here is status and tasks with PID1 and overclocked board- just for your :
# tasks
Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 0 ms
1 - GYRO/PID, max = 377 us, avg = 235 us, total = 5331 ms
3 - SERIAL, max = 65 us, avg = 2 us, total = 4 ms
4 - BEEPER, max = 9 us, avg = 0 us, total = 1 ms
5 - BATTERY, max = 40174 us, avg = 1 us, total = 42 ms
6 - RX, max = 114 us, avg = 74 us, total = 86 ms
12 - TELEMETRY, max = 45 us, avg = 2 us, total = 23 ms

# status
System Uptime: 13 seconds, Voltage: 152 * 0.1V (4S battery - OK), System load: 0.08
CPU Clock=84MHz, GYRO=MPU6050
Cycle Time: 493, I2C Errors: 0, config size: 2020

borisbstyle, 2016-01-01:
yeah pid1 speed is significantly faster. No even need for overclocking it.

How does luxfloat look like overclocked?

carebear482, 2016-01-01:
Luxfloat with set emf_avoidance = on :
Task list:
0 - SYSTEM, max = 15 us, avg = 0 us, total = 241 ms
1 - GYRO/PID, max = 463 us, avg = 347 us, total = 1868349 ms
3 - SERIAL, max = 61457 us, avg = 2 us, total = 1355 ms
4 - BEEPER, max = 16 us, avg = 0 us, total = 538 ms
5 - BATTERY, max = 40172 us, avg = 1 us, total = 751 ms
6 - RX, max = 119 us, avg = 77 us, total = 18920 ms
12 - TELEMETRY, max = 38 us, avg = 3 us, total = 4903 ms

# status
System Uptime: 3721 seconds, Voltage: 0 * 0.1V (0S battery - NOT PRESENT), System load: 0.22
CPU Clock=84MHz, GYRO=MPU6050
Cycle Time: 497, I2C Errors: 0, config size: 2020

borisbstyle, 2016-01-01:
so it does arm like that now?

carebear482, 2016-01-01:
Unfortunately no, I'll stick with PID1 then. You should warn rev5 users that thing like that can happen with luxfloat.

borisbstyle, 2016-01-01:
It was mentioned already in rcg discussions and I did some improvements to
increase the chance that it works. And it works now, but there are
apparantly slight hardware differences and features that people use so its
not 100% guarantee that it will work.
For example it would work on Pwm and ppm but not sbus for you.

pid1 just flies fine anyway.

carebear482, 2016-01-01:
Yep, no problem. Keep up good work ;)

borisbstyle, 2016-01-04:
I still wonder why it didnt work in luxfloat. From seeing the task performance it should be able to do it.
I will check it soon on my naze boards again.

borisbstyle, 2016-01-08:
@carebear482 
Did you btw try without telemetry?

carebear482, 2016-01-08:
I didn't tried without telemetry. Telemetry would be too big loss for me.

carebear482, 2016-01-08:
It seems that 2 khz and floating points are too much for older rev5 naze boards and since it seems that f1 does better job with PID1 controller then I don't see any point to even use LUX on it or am I missing something?

borisbstyle, 2016-01-08:
rewrite does fantastic job on 2khz! With faster rates you get more precision in PID's due to small steps. That basically even removes the need for floating point math.

lazd, 2016-01-14:
I'm having the same problem as of Betaflight 2.2.0.

## Problem

The board does not arm with stick commands or a switch. Instead the green LED flashes. The board may arm intermittently.

## Setup

**Board**: Naze rev6b
**Versions**: Betaflight 2.2.0 and 2.3.0

Partial dump (just the important stuff that differs from the defaults):

```
# features
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125
feature BLACKBOX

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 0 115200 57600 0 115200
set serialrx_provider = SBUS

# Telemetry
set telemetry_inversion = ON

# Blackbox
set blackbox_device = SPIFLASH

# PID Controller
set pid_controller = LUX
```

## Workarounds

1. Use Betaflight 2.1.6.

2. In Betaflight 2.2.0 and 2.3.0, disable the accelerometer with `set acc_hardware = 1`.

## Notes

`tasks` output on 2.3.0:

```
# tasks
Task list:
0 - SYSTEM, max = 12 us, avg = 0 us, total = 4 ms
1 - GYRO/PID, max = 898 us, avg = 775 us, total = 50736 ms
2 - ACCEL, max = 153 us, avg = 122 us, total = 5301 ms
3 - SERIAL, max = 3094 us, avg = 2 us, total = 18 ms
4 - BEEPER, max = 13 us, avg = 0 us, total = 11 ms
5 - BATTERY, max = 132 us, avg = 1 us, total = 15 ms
6 - RX, max = 211 us, avg = 178 us, total = 550 ms
8 - BARO, max = 134 us, avg = 71 us, total = 345 ms
10 - ALTITUDE, max = 306 us, avg = 174 us, total = 348 ms
12 - TELEMETRY, max = 11 us, avg = 0 us, total = 46 ms
```

diggidydaz, 2016-01-15:
I had the same issue today and was able to arm in LuxFloat after overclocking.
I am running a DragonFly32 (Naze target)

Here is my status and tasks.  I dont understand them yet lol.  But I'm waiting for my F3 board to arrive so that I can compare.

Thanks Boris for the great software

# status
System Uptime: 224 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.29
CPU Clock=80MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 988, I2C Errors: 0, config size: 2020

# tasks
Task list:
0 - SYSTEM, max = 9 us, avg = 0 us, total = 16 ms
1 - GYRO/PID, max = 1023 us, avg = 615 us, total = 173265 ms
2 - ACCEL, max = 144 us, avg = 113 us, total = 31642 ms
3 - SERIAL, max = 59155 us, avg = 2 us, total = 610 ms
4 - BEEPER, max = 7 us, avg = 0 us, total = 35 ms
5 - BATTERY, max = 103 us, avg = 1 us, total = 48 ms
6 - RX, max = 59202 us, avg = 50 us, total = 1050 ms
7 - COMPASS, max = 147 us, avg = 114 us, total = 367 ms
8 - BARO, max = 130 us, avg = 99 us, total = 3046 ms
10 - ALTITUDE, max = 244 us, avg = 136 us, total = 1498 ms
13 - LEDSTRIP, max = 340 us, avg = 34 us, total = 1423 ms

SoulPictures, 2016-01-15:
You got the same as the guy some posts higher, your total calculations take on average 1164µs which is about 1,16ms, which would limit you to 850 Hz instead of 1kHz (1000Hz) you need to disable features until your average is a good bit below 1000. so you need to reduce by around 170 (disable Compass and Baro and it could be enough that you can arm)

diggidydaz, 2016-01-15:
How do I go about disabling those two?  Can I do it from the CleanFlight CLI?
I am able to arm and fly now, but I am sure that my FC would perform much better with unnecessary tasks terminated. 

Thanks

carebear482, 2016-01-15:
Write this to CLI, it will disable unnecessary things :)
set acc_hardware = 1
set baro_hardware = 1
set mag_hardware = 1
save

borisbstyle, 2016-01-15:
Some tasks do not run each loop guys. It is not just sum of all average.

load is the indication. load should not reach value if 1

borisbstyle, 2016-01-20:
Closing this issue. It should work better in 2.3.3 anyway

closed 2016-01-20 by borisbstyle

carebear482, 2016-01-20:
But still rewrite outpeform lux on f1 boards, right?

borisbstyle, 2016-01-20:
of course.....but just saying that lux should now be flyable

carebear482, 2016-01-20:
Thanks, good to know. One last question and then we can close this thread. Will fast pwm and changing motor_pwm rate to 4000 will make anything better or rather worse? Since it loses sync with gyro but makes things faster. Just a question what is bothering me few days.

borisbstyle, 2016-01-20:
motor pwm 4000 would possibly be unstable as oneshot125 wouldnt have enough
space between pulses.

But anyway it wont make anything better technically. update is still the
same but now you get the chance  that you skip loop update

carebear482, 2016-01-20:
And what about fast pwm? Will it improve oneshot?

borisbstyle, 2016-01-20:
Its the same just not synced to loop.

RiceCakeWtf, 2016-02-13:
There's a potential for this still being an issue on recent builds (2.4.0-RC7). I run on an SPrF3 board. LuxFloat, GPS, 2khz, full Blackbox logging, causes sporatic arming issues. Often the craft will arm, fly, disarm, and fail to arm again until rebooted. Disabling some treats resolves the issue. Just posting so anyone observing this is aware -- it may be possible to "top out" the capability of even an F3 board with recent revisions.

borisbstyle, 2016-02-13:
did you calibrate acc on your spfracing board when it doesnt arm?

RiceCakeWtf, 2016-02-13:
The ACC is calibrated as the craft arms fine initially, flies normally, and lands normally. Landing, disarming, and not even moving the craft seems to cause it to be unable to re-arm. Will give this a quick test just to verify and will try recalibrating the accelerometer if it doesn't.

borisbstyle, 2016-02-13:
just recalibrate with sticks om the ground. left stick left up and right
stick down i believe.

Sometimes acc can show drift and especially if it is calibrated on the edge
of the small angle limit.

RiceCakeWtf, 2016-02-13:
Yep, just tried again. Power up, arm, motors idle, disarm, arm (successfully), motor idle, disarm, then nothing. Its fairly hit-or-miss, I've had it never arm, I've had it arm several times fine. When it refuses to arm though, its DOA, nothing functions anymore.

I did run the accelerometer calibration stick combo, though, as I understand entirely where you're coming from. Still, it doesn't seem to have any effect on it.

Interestingly though, I plugged it in just now and left it sit a few minutes before trying to arm it. It failed to arm. Hooking it up with a USB cable, the device was enumerated in Windows and the Configurator connected to the virtual COM port fine...but the connection eventually failed with 'no data received'.

`
Entering CLI Mode, type 'exit' to return, or 'help'

# dump master

# version
# BetaFlight/SPRACINGF3 2.4.0 Feb  9 2016 / 12:23:04 (dd2478f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature GPS
feature FAILSAFE
feature RX_PARALLEL_PWM
feature ONESHOT125
feature BLACKBOX

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 115200 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 0,0:::0
led 1 0,0:::0
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set emf_avoidance = OFF
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set rc_smoothing = OFF
set roll_yaw_cam_mix_degrees = 0
set min_throttle = 1080
set max_throttle = 1864
set min_command = 900
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set enable_fast_pwm = OFF
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_config = ON
set gps_auto_baud = ON
set gps_wp_radius = 200
set nav_controls_heading = ON
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set serialrx_provider = SPEK1024
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_pid_compensation = OFF
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 700
set gyro_lpf = OFF
set gyro_sync_denom = 4
set gyro_soft_lpf =  60.000
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 5
set yaw_deadband = 7
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_control_direction = 1
set yaw_motor_direction = 1
set airmode_saturation_limit = 50
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set gimbal_mode = NORMAL
set acc_hardware = AUTO
set acc_lpf_hz = 20
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set baro_hardware = AUTO
set mag_hardware = AUTO
set mag_declination = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SPIFLASH
set beeper_off_flags = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# `

I'll simply go back to 1khz. It worked perfectly anyhow, was just experimenting, figured it might help to share this curious issue I was having.

borisbstyle, 2016-02-13:
to me it all sounds like acc....especially when you have sitting it on the
ground.

Its without doubt acc for sure.

borisbstyle, 2016-02-13:
increase small_angle to something bigger

Also do a status dump to check cpu usage.

RiceCakeWtf, 2016-02-13:
`Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 5 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:14%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 500, I2C Errors: 0, config size: 1324

# status
System Uptime: 33 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:12%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 496, I2C Errors: 0, config size: 1324

# status
System Uptime: 48 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:33%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 493, I2C Errors: 0, config size: 1324

# status
System Uptime: 58 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 494, I2C Errors: 0, config size: 1324

# status
System Uptime: 60 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 497, I2C Errors: 0, config size: 1324

# status
System Uptime: 61 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 585, I2C Errors: 0, config size: 1324

# status
System Uptime: 62 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 490, I2C Errors: 0, config size: 1324

# status
System Uptime: 62 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 490, I2C Errors: 0, config size: 1324

# status
System Uptime: 65 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:100%
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Cycle Time: 488, I2C Errors: 0, config size: 1324

# `

Attempted to arm, got %100 CPU, motors never came up.

Going back to 1khz; it solves the problem completely.

borisbstyle, 2016-02-13:
100% cpu doesnt allow you to arm.

check tasks to see what task is causing high cpu

richsemc, 2016-03-12:
I have the same arming problem,(spracingF3) everything working fine but will not arm, when I arm I get the bleep to say armed but nothing on motors, all motors test out ok, sonar,gps,telemetry disabled