betaflight/betaflight #95: Hard fault on STM32F3 Discovery Board

tannewt, 2016-01-03:
My STM32F3 Discovery board isn't working with betaflight and I think I've narrowed it down to the gyro->intStatus not being set for the acc/gyro L3GD20. I used gdb to step all the way to the "gyro->intStatus(&mpuDataStatus)" line in gyro_sync.c. Stepping from there leads to the hard fault. How is the gyro sync feature supposed to working with other gyros (including fake gyro)? I'm happy to fix it but I'm not certain how.

tannewt, 2016-01-03:
I also confirmed this by replacing the `gyroSyncCheckUpdate()` call in `taskMainPidLoopCheck` in mw.c with `false` so that it falls back to the looptime condition and it runs fine now.

henn1001, 2016-01-03:
I think it is not supported at the moment.

borisbstyle, 2016-01-03:
yes discovery is not in the list of supported targets.

What you do is just bypassing the gyro sync feature instead of solving it
by inplementing the right interrupt handles for it.

tannewt, 2016-01-03:
Should I add a macro to do the bypass? Is there a list of supported targets
somewhere I missed?

borisbstyle, 2016-01-04:
The ones in the release list:
https://github.com/borisbstyle/betaflight/releases

You can fix the target if there is INT pin connected. Just define what pin and try to fix gyro sync

closed 2016-01-04 by borisbstyle