betaflight/betaflight #96: Motors gradually increase to full throttle when throttle stick is at min throttle

Bassbitz, 2016-01-03:
upon arming (motor stop)
when raising throttle just bit on the sticks
the motors would continue to go to max despite the throttle input i gave after that.

on the motor test, it works as it should.

Naze32 rev 5.

Bassbitz, 2016-01-03:
Video footage go the problem
[Video.MOV.zip](https://github.com/borisbstyle/betaflight/files/76542/Video.MOV.zip)

borisbstyle, 2016-01-03:
are you trying in air mode?

Bassbitz, 2016-01-03:
[image]

With airmode turn on or off, it gave me the same results.

Bassbitz, 2016-01-03:
Tested with motor stop off, idle throttle works normally. Giving any throttle input has the same issue..

borisbstyle, 2016-01-03:
if you are testing in air mode that is a normal behaviour. Without air mode
it should not happen.

In air mode....once you move stick above min check for 1 second or arm with
motor stop the copter will think its flying.

Once you take off it will all be fine. On the ground it is not able to
correct so the error will be building up

Bassbitz, 2016-01-03:
will head out and test out with props to verify this.
But at the moment airmode is turned off and it is still behaving as such.

SoulPictures, 2016-01-03:
If i get everything right the behavior seems normal, as soon as you apply enough throttle the PID-Controller will try to keep you Quad level (if in horizon mode) or in its current position (acro mode). Due to measuring inaccuracies (gyro and acc drift/offset) your quad will slowly build up power to keep the desired configuration. If you have props on the quad it would simply lift up a bit (on one side) and roughly stay there

Bassbitz, 2016-01-03:
Tested it just now and the throttle definitely has something wrong. The only way to even hover was to go to air mode, which behaved differently but still badly.

I'm going to refresh this version again and see if that helps if now I'm going back an older ver of betaflight. 

ill drop my CLI dump here if it helps

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# BetaFlight/NAZE 2.2.0 Dec 31 2015 / 00:50:34 (560bd74)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature FAILSAFE
feature RX_PARALLEL_PWM
feature ONESHOT125

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
[... 244 lines cut ...]
set p_level = 20
set i_level = 20
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set gyro_soft_lpf = ON
set dterm_cut_hz = 0
set yaw_pterm_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 65
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 80
set tpa_rate = 10
set tpa_breakpoint = 1550

borisbstyle, 2016-01-04:
This is a normal behaviour when throttle stick above min_check as the copter will try to correct itself.

Put the stick below min_check and it will stop spooling up and slowly spool down

closed 2016-01-04 by borisbstyle

dv350, 2016-03-13:
@Bassbitz  : I'm facing the exact same issue. When I used cleanflight's motor tab and increase throttle everything looks good but when I use my futaba and leave at 10 % throttle, the motors automatically keep going up until it reachs full throttle. When I try to fly it, after hovering for a second or two it flips. I don't know if its the new cleanflight or betaflight. One thing I did different in this build was I used the power osd pro. The 5v connection to power the fc was used and for ground i connected it to the ESC ground. Please tell me you found a solution ?

borisbstyle, 2016-03-13:
that is how airmode works....when on ground and your stick is
above.min_check  the error will keep growing as it thinks it is flying

dv350, 2016-03-13:
Well I'm not on airmode, I don't use airmode until I get my quad flying
fine. When I try it hover it, it feels strange and then it flips.
Don't know the issue here is, faulty naze maybe.

grymlock13, 2016-03-29:
Hi guys!  Just wondering if you figured out a fix for this.  I just flashed over to 2.5.4 and am encountering the exact same issue.  It's strange because on my other 250mm quad, which is almost an identical build save the escs (Rotorgeeks 20a vs. Kiss 18s); The kiss18s one is flying like a dream.

dv350, 2016-03-29:
Hey,

It was a bad naze, I changed it and it worked perfectly.
Good luck!

- Monish

grymlock13, 2016-03-29:
Haha, dang. Not the answer I hoped for but at least it's an easy fix! =)  Alrighty roo!  Thanks a ton man!  

-Ed

rayone, 2016-04-21:
I have an ImmersionRC Vortex Pro 250 (IRCFusionF3) which has just started exhibiting this issue on 2.6.0 and 2.6.1.
I'm also unable to revert to previous know good 2.5.3.
I get a bad flash when verifying in CFC.
The issue is not present on 2 other Naze32 quads.

eloygambin, 2016-04-23:
Hello guys! Have the same problem.... 

When I use horizon/angle in Luxfloat the increase is slowly, but using rewrite is crazy!! very fast!

Did find a solution?

borisbstyle, 2016-04-23:
Just keep throttle below min_check and keep roll and pitch centered and it
will not spool up unless you have very bad oscillations ;)

eloygambin, 2016-04-23:
Boris, I tried with propellers and the quads is going good.... but When I try on the table (no propellers) is when I had the problems....It's so weird!

RC-Slater, 2016-04-23:
Please stop asking this question. It's not weird. All is well. https://github.com/borisbstyle/betaflight/wiki/Frequently-Asked-Questions#why-do-my-motors-keep-accelerating-on-the-bench-when-i-arm-without-props-

petrposvic, 2018-05-05:
I have the same problem. Solved by PID config. Start with P default (i.e. 40), I zeros, D zeros and watch what happens.

brandonsD, 2018-08-10:
Boris how do I pm you on here??

DieHertz, 2018-08-10:
You don't PM Boris. What do you want?

brandonsD, 2018-08-10:
I am having an issue with my setup. I have a qav250 with cc3d and r9d receiver. I’m getting a problem with the yaw applied to the right will incline throttle up and turn right and yaw left will decent and turn left loose power in other words. Idk what’s going on or how to diagnose this. Thank if you can help