betaflight/betaflight #97: CC3D automáticamente flips

alvarobeiste, 2016-01-03:
I have used BetaFlight succesfully in the last two weeks, my quad flew great with this firmware in its CC3D, but this week I broke two quads. The issue is that the quad randomly starts an infinite flip into the ground. I thought it was a faulty CC3D, but I replaced it with a new one and the problem is still there. I have ordered a Naze to test if the issue is gone with this controller, but I'm still unconfident to flash the new BetaFlight into it. Do you think that a old version would work? Before I updated last week, everything worked perfectly...

alvarobeiste, 2016-01-03:
Sorry, I was trying to write "automaticly", but the spanish corrector in my phone changed it :)

borisbstyle, 2016-01-04:
Are you sure it is not your ESC's?

There is a seperate issue about CC3D issues. 
https://github.com/borisbstyle/betaflight/issues/64

closed 2016-01-04 by borisbstyle