betaflight/betaflight #99: Reverse motor direction (quad)

bbbenji, 2016-01-07:
I am looking to reverse the motor direction as it would be a lot of trouble to resolder. Doing some searching I ran across the command `yaw_direction`, which when set to -1 does the trick. Now it seems to be `yaw_motor_direction` but this does not do anything on a quad as far as I can tell. 

Is there a simple way of changing the motor direction in betaflight?

NightHawk32, 2016-01-07:
I don't really understand what you're trying to do there but if you're really looking for a way to reverse the rotation direction of a motor, I can tell you there isn't one in betaflight. Thats all up to your ESCs, you could use the passthrough for BLheli ESCs to reverse the direction in there. But the fastest way would be to just flip those two wires on the motor.

borisbstyle, 2016-01-07:
@bbbenji 
That's an ESC function. Cant be done in the FC

closed 2016-01-07 by borisbstyle

ProfX-FPV, 2017-02-27:
@borisbstyle 

Boris,

agreed on the ESC function with brushless motors. But how do we handle brushed setups like the Tiny Whoop?

The expected motor layout / direction in Betaflight is 
- Front left: Clockwise; front right: counter clockwise;
- Back left: Counter clockwise; back right: clockwise.

The original Blade Inductrix motor layout is
- FL: CCW; FR: CW;
- BL: CW; BR: CCW.

Now people swapping the Inductrix board with one of the Betaflight compatible boards (Beebrain F1, Beecore F3, Acrowhoop F3 ...) will have to swap motors too to match Betaflight's layout. That's what I had to do (and it works fine, BTW).

Some argue now that it would be preferable to keep the original Inductrix layout because of less dirt wash on the camera, the thrust isn't working against the direction of the vanes of the original Inductrix frame and other issues involving yaw on a quad with a very short motor distance. 

Brushed motors are not supposed to run backwards - short motor life, bad efficiency.

I understood that this might be done by combining "yaw_motor_direction = -1" with a custom motor mix, but I am not sure, and never got it working.

What would be your approach to solve this in a brushless config running Betaflight 3.1.x?

Cheers and keep up your great work - I really really appreciate the effort!

AndersHoglund, 2017-02-27:
Keep motors as they are, only thing needed is a CLI command:

    set yaw_motor_direction = -1
    save

I have this on all my rigs, to throw dirt outwards.
NB, this still requires the default betaflight motor order! I.e:
FL: 4 FR: 2
BL:3 BR: 1

If OP or RF motor order is used, try the "QuadX 1234" mixer.

ProfX-FPV, 2017-02-28:
@AndersHoglund thanks, that did it.

I switched motors on a Tiny Whoop / BeecoreF3 / Betaflight 3.1.6 from Betaflight layout to Inductrix layout (moving them by 90°) and set yaw_motor_direction = -1. Flies fine. 

Do I understand correctly that if I wanted to get the same setup on my brushless quads I would reverse motor direction using BlHeli on all 4 ESCs (or rewire them) and use "set yaw_motor_direction = -1" as well?

And just for curiosity: what does "OP or RF Motor order" stand for?

mikeller, 2017-02-28:
OP: OpenPilot, RF: RaceFlight, motors are numbered like this: https://www.rcgroups.com/forums/showpost.php?p=33003889&postcount=5