betaflight/betaflight #101: 2.2.0 Gtune won't work

wavecut, 2016-01-09:
Hi, I'm flying the 2.2.0 firmware on Naze32, and I found the Gtune mode didn't work and all the P value didn't changed after gtune.

After compared the source code with 2.1.6(Gtune works well in this version), it's seems like 2.2.0 version didn't call the updateGtuneState() function.

The following code present in 2.1.6 but missed in 2.2.0
--------------------------
#ifdef GTUNE
        updateGtuneState();
#endif
--------------------------

hope it can be fixed in next Revision.

Thanks for your awesome firmware, I'm enjoy in it.

--
wavecut@qq.com
2016/1/10

carebear482, 2016-01-10:
I can confirm this.

borisbstyle, 2016-01-11:
This will be fixed in this PR
https://github.com/borisbstyle/betaflight/pull/104

commit 886d2f09 (2016-01-12): Fix #101: Add updateGtuneState() function call
commit bb5a3283 (2016-01-12): Merge pull request #104 from wavecut/betaflight

Fixed #101: Add updateGtuneState() function call
closed 2016-01-12 by borisbstyle