betaflight/betaflight #102: New feature- motor stop when armed and motor stop off when on airmode

carebear482, 2016-01-10:
Is it possible to add feature that has motor stop enabled when in arm mode and motor stop disabled when using airmode? I like airmode a lot, but I hate to look for disarm switch when flying in tight spots.

borisbstyle, 2016-01-10:
having motor stop in air mode kind of counteracts with the way how air mode works.

Just get used to disarm switch .....it works great!

borisbstyle, 2016-01-10:
btw air mode works with motor stop! Only the motors stop with 0 throttle

closed 2016-01-10 by borisbstyle

carebear482, 2016-01-11:
My point was just reverse- to have motor stop on normal rate mode and motor stop disabled in air mode. It would be fantastic if you could make motor stop as a mode in Modes tab so everyone can set it up via transmitter switch like needed.

Teracis, 2016-01-13:
Carebear482 the idea you are asking for can be easily achieved already. I don't think it has been merged yet but dropax has a version with idle on mode. This lets you have motor_stop enabled and then activate idle on via a switch which makes it work like motor_stop is disabled. You could set up idle on mode identically to your airmode and then you wouldn't stop motors during airmode. 

Good luck with it and happy flying!