betaflight/betaflight #107: Naze 32 LED_STRIP Feature not able to enable on Version 6B boards

Vanbladez, 2016-01-12:
Hi Boris, everyone.

Building a new Raceblade and looking to setup the NAZE32 with a couple of LED Bars running just under 5v. Issues found with the betaflight 2.2.0 when trying to enable the LED_STRIP Feature and save.

Settings unable to stay active and disables automatically.

Same board loaded with Clean flight latest (1.11.0)

Is there a know problem, issue for enabling the LED functions on Beta flight?

Would appreciate feedback or if im doing something wrong in the setup, additional sets needed?

Cheers Vanblade.

borisbstyle, 2016-01-12:
Are you running in 2khz mode?

Vanbladez, 2016-01-12:
Ok, Ive found a bit more info. After reflashing back to your latest 2.2.0 version and checking setup again.

Im running 2khz mode and the LEDS are working okay.

However, turning on soft serial switch for telemetry outputs seems to stop the LEDS working again.
Here is the status & Dump.

Is this correct, to be able to have soft serial working along with the LED Strip.

Cheers Van

# status
System Uptime: 11 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.07
CPU Clock=72MHz, GYRO=MPU6500
Cycle Time: 496, I2C Errors: 0, config size: 2020

# tasks
Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 1 ms
1 - GYRO/PID, max = 315 us, avg = 219 us, total = 7014 ms
3 - SERIAL, max = 15648 us, avg = 2 us, total = 23 ms
4 - BEEPER, max = 11 us, avg = 0 us, total = 2 ms
5 - BATTERY, max = 119 us, avg = 1 us, total = 3 ms
6 - RX, max = 80 us, avg = 49 us, total = 54 ms
12 - TELEMETRY, max = 8 us, avg = 0 us, total = 16 ms
13 - LEDSTRIP, max = 410 us, avg = 31 us, total = 65 ms

#  
# dump

# version
# BetaFlight/NAZE 2.2.0 Jan  7 2016 / 01:06:58 (cad401d)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature VBAT
feature FAILSAFE
feature TELEMETRY
feature LED_STRIP
feature ONESHOT125

# map
map AETR1234
[... 269 lines cut ...]
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set gyro_soft_lpf = ON
set dterm_cut_hz = 0
set yaw_pterm_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

borisbstyle, 2016-01-13:
what do you mean it doesnt enable?
What procedure do you follow?

husyildiz, 2016-01-13:
+1 I have same problem with flip32 10dof softserial and led strip don't
work at the same time. When enable softserial led stop disabled
automatically.

borisbstyle, 2016-01-13:
I will check that soon.

mikeller, 2016-01-13:
According to the [documentation](https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20Naze32.md), LED_STRIP and SOFTSERIAL are mutually exclusive (because they share the same pins), which would explain @Vanbladez ' problems.

husyildiz, 2016-01-13:
How can we split ?

husyildiz, 2016-01-13:
How can we split ? Beacuse of when we want to use telemetry via softserial same time led strip.

borisbstyle, 2016-01-13:
guys are you using leds on softserial connection?

husyildiz, 2016-01-13:
Ch5 for led .

Vanbladez, 2016-01-13:
Currently using pin 5 for leds and pin 6 for output telemetry with soft serial.  Is it changing to uart?

---
https://github.com/borisbstyle/betaflight/issues/107#issuecomment-171207545

mikeller, 2016-01-13:
Guys there is a discussion of why this **will not** work on Naze32 and derivatives: https://github.com/cleanflight/cleanflight/issues/465 Please go read that. As an option I would recommend moving telemetry to USART1 or USART2. This will also work nicely if the port is shared with MSP/CLI (for cleanflight configurator), with the FC switching between the two when arming / disarming.

borisbstyle, 2016-01-13:
5 is UART2.... so it is not softserial and it doesnt work till you enable
softserial ....do I got that right?

borisbstyle, 2016-01-13:
telemetry cannot run on UART2. It needs inversion

Vanbladez, 2016-01-13:
Thanks chaps for the soft serial confirmation.  Looks like I was mixing up a bit.

Will look to try the Uart 2 for pin 3 or 4 output on the telemetry and inverted,  and enable the Leds strip for pin 6.

Thanks for your help so far

Cheers Van

---
https://github.com/borisbstyle/betaflight/issues/107#issuecomment-171209769

husyildiz, 2016-01-13:
Thanks 

Must read more I think.

Vanbladez, 2016-01-13:
I'll let you know how it goes when tested but delayed now today / tomorrow =  travelling

https://github.com/borisbstyle/betaflight/issues/107#issuecomment-171214630

husyildiz, 2016-01-13:
Waiting to hear good news :)

mikeller, 2016-01-13:
@borisbstyle is right: If using a FrSky RX, then the telemetry signal needs to be inverted (availabe on the TXI pin for USART1) . For other RX, non inverted signal and USART1 or USART2 might be an option.

borisbstyle, 2016-01-13:
I will close this issue since it isnt an issue at all ;)

closed 2016-01-13 by borisbstyle

husyildiz, 2016-01-13:
Find solution

http://www.rcgroups.com/forums/showthread.php?t=2389724

@borisbstyle is right :)

@Vanbladez