betaflight/betaflight #110: Naze32 rev6 won't arm with accelerometer enabled as of 2.2.0

lazd, 2016-01-14:
## Problem

The board does not arm with stick commands or a switch. Instead the green LED flashes. The board may arm intermittently if you keep trying.

## Setup

**Board**: Naze rev6b
**Versions**: Betaflight 2.2.0 or 2.3.0

Partial dump (just the important stuff that differs from the defaults):

```
# features
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125
feature BLACKBOX

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 0 115200 57600 0 115200
set serialrx_provider = SBUS

# Telemetry
set telemetry_inversion = ON

# Blackbox
set blackbox_device = SPIFLASH

# PID Controller
set pid_controller = LUX
```

## Workarounds

1. Use Betaflight 2.1.6.

2. In Betaflight 2.2.0 and 2.3.0, disable the accelerometer with `set acc_hardware = 1`.

## Notes

`tasks` output on 2.3.0:

```
# tasks
Task list:
0 - SYSTEM, max = 12 us, avg = 0 us, total = 4 ms
1 - GYRO/PID, max = 898 us, avg = 775 us, total = 50736 ms
2 - ACCEL, max = 153 us, avg = 122 us, total = 5301 ms
3 - SERIAL, max = 3094 us, avg = 2 us, total = 18 ms
4 - BEEPER, max = 13 us, avg = 0 us, total = 11 ms
5 - BATTERY, max = 132 us, avg = 1 us, total = 15 ms
6 - RX, max = 211 us, avg = 178 us, total = 550 ms
8 - BARO, max = 134 us, avg = 71 us, total = 345 ms
10 - ALTITUDE, max = 306 us, avg = 174 us, total = 348 ms
12 - TELEMETRY, max = 11 us, avg = 0 us, total = 46 ms
```

readerror67, 2016-01-14:
Have you tried without telemetry?

lazd, 2016-01-14:
@readerror67 I just tried disabling SOFTSERIAL and TELEMETRY and it still won't arm.

Output of `tasks`:

```
# tasks
Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 1 ms
1 - GYRO/PID, max = 930 us, avg = 785 us, total = 21071 ms
2 - ACCEL, max = 152 us, avg = 121 us, total = 2025 ms
3 - SERIAL, max = 147 us, avg = 2 us, total = 8 ms
4 - BEEPER, max = 9 us, avg = 0 us, total = 3 ms
5 - BATTERY, max = 40176 us, avg = 1 us, total = 45 ms
6 - RX, max = 178 us, avg = 146 us, total = 283 ms
8 - BARO, max = 133 us, avg = 70 us, total = 120 ms
10 - ALTITUDE, max = 273 us, avg = 157 us, total = 122 ms
```

lazd, 2016-01-14:
However, when I disable the barometer with `set baro_hardware = 1` AND disable telemetry, it will arm.

It will not arm with only the barometer disabled and telemetry/softserial enabled.

SoulPictures, 2016-01-14:
to check if it even possible to arm with all the features you have enabled on that target you should sum up all AVG values from the task list. 

If i do the sum for your task list i get: ~1300 so you need to disable stuff till you get lower the let's say 950 to be able to run 1kHz mode.

borisbstyle, 2016-01-14:
You are using luxfloat on F1 boards, which is also much slower and that becomes an issue with so many features activated.
If you want to keep all features just go for rewrite and otherwise disable something. Baro is the main problem.

lazd, 2016-01-14:
@borisbstyle understood. However, this doesn't explain why the same setup worked just fine in Betaflight 2.1.6 before the new task scheduling method.

I just tested everything under 2.1.6, and it appears to work -- I get telemetry back on my Taranis with the barometer's reading and the accelerometer works too. However, loop times do jump around from between 966 and 1160. Disabling the barometer does not affect this behavior, however, when I disabled the accelerometer I get closer to 1000 much more consistently, between 994 and 1005.

Is it that I was really overloading the CPU all along, resulting in inconsistent looptimes, and the arming issue was just a function of the new task scheduler recognizing that?

borisbstyle, 2016-01-14:
the new firmware has protection what prevents you arming when you cpu
doesnt act healthy.
2.1.6 didnt have that feature.

can you type status and see how much mhz your board is.

Also try emf_avoidance = on.  Maybe it will boost it a bit.

lazd, 2016-01-14:
Here you go:

```
# status
System Uptime: 20 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 995, I2C Errors: 0, config size: 2020
```

This is with barometer and accelerometer enabled on 2.1.6.

I will try `emf_avoidance = on` and report back.

lazd, 2016-01-14:
I just tried `emf_avoidance = on` and though it does seem to help, I am still seeing inconsistent looptimes that range between 990 and 1080.

I guess it's time for a F3 board! Thanks for the help.

I suppose we can close this issue as it's clear the arming problem was due to giving the poor little F1 too much work.

lazd, 2016-01-14:
Quick update: I was able to get things working in Betaflight 2.3.x by disabling the barometer with `set baro_hardware = 1` and enabling EMF avoidance with `set emf_avoidance = on`. This results in the following:

```
# status
System Uptime: 36 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 0.63
CPU Clock=84MHz, GYRO=MPU6500, ACC=MPU6500
Cycle Time: 988, I2C Errors: 0, config size: 2020
```

```
# tasks
Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 3 ms
1 - GYRO/PID, max = 1097 us, avg = 701 us, total = 36222 ms
2 - ACCEL, max = 144 us, avg = 113 us, total = 6266 ms
3 - SERIAL, max = 16860 us, avg = 1 us, total = 53 ms
4 - BEEPER, max = 9 us, avg = 0 us, total = 8 ms
5 - BATTERY, max = 111 us, avg = 1 us, total = 9 ms
6 - RX, max = 178 us, avg = 147 us, total = 420 ms
12 - TELEMETRY, max = 9 us, avg = 0 us, total = 51 ms
```

Like @borisbstyle said, the poor F1 just couldn't handle all the features. The reason it worked in 2.1.6 was because there was no protection for an unhealthy overloaded CPU. I'm going to close this issue as we know what the problem is and how to solve it.

closed 2016-01-14 by lazd