betaflight/betaflight #111: Overloaded/Unhealthy CPU makes it impossible to connect to Cleanflight configurator

lazd, 2016-01-14:
## Problem

On F1 boards that aren't able to keep up with the enabled features, it sometimes becomes impossible to connect to Cleanflight.

## Setup

**Board**: Naze rev6b
**Versions**: Betaflight 2.2.0 or 2.3.0

Partial dump (just the important stuff that differs from the defaults):

```
# features
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125
feature BLACKBOX

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 0 115200 57600 0 115200
set serialrx_provider = SBUS

# Telemetry
set telemetry_inversion = ON

# Blackbox
set blackbox_device = SPIFLASH

# PID Controller
set pid_controller = LUX
```

## Workarounds

1. Flash the board again (bridging the bootloader pins) and don't enable as many features

## Corroborating reports

[TwinTurbo95 on RCGroups](http://www.rcgroups.com/forums/showpost.php?p=33716692&postcount=14201)

lazd, 2016-01-14:
Update: I'm actually now having problems connecting even after flashing.

borisbstyle, 2016-01-14:
If you really want to use every possible feature there is you really need to get rid of naze32  or other F1 board. Get a F3 FC or start flying rewrite.

You can also disable accelerometer btw.

lazd, 2016-01-14:
@borisbstyle I totally understand, I've got some Dodos on the way. However, it seems there is something going wrong internally that makes it impossible to connect to reconfigure.

borisbstyle, 2016-01-14:
It works for me, but I dont use that many options. Serial connection doesnt have high priority when things get overloaded, but I am still able to connect and do everything. 
What PC are you using?

lazd, 2016-01-14:
Unfortunately the problem is intermittent I was able to connect after I flashed a few more times. It is possible that there is something wrong with the board, I will continue to try to reproduce it consistently.

I've tried this both on a Macbook Pro running Windows 7 and a custom build desktop running OS X 10.10.

Vincka, 2016-01-17:
Hello,

Board : **SP Racing F3 Deluxe**
Betaflight version : **Betaflight 2.3.1**

The same problem for me with SP Racing F3:

- STM32F303 chip very hot. Impossible to keep finger on it
- No more blue led light on
- Impossible to connect to CleanFlight Configurator
- Impossible to flash, even with boot jumper bridged

Here is Cleanflight dump, just before upgrading from 2.1.6 to 2.3.1.
[Betaflight_2.1.6_dump_2015-01-20.txt](https://github.com/borisbstyle/betaflight/files/93537/Betaflight_2.1.6_dump_2015-01-20.txt)

borisbstyle, 2016-01-17:
That sounds like you have other problem with your power supply. There is no
way the chip should get hot :)

Vincka, 2016-01-17:
Hello,

No, chip very hot also when only connected to my computer through USB port. I plug the MicroUSB, and about 5 seconds later, chip is hot enough to make impossible to keep finger on it...

Many thanks to you for the great work you do with BetaFlight :).

borisbstyle, 2016-01-17:
maybe you have a short. Firmware cant do that.
Maybe its the little 5V regulator. I once had the same issue a while ago.
It is just below the chip. Little black thing.

lazd, 2016-01-17:
I've seen the hot chip problem before and it's been a bad board. If you want to triple check, you can load up Cleanflight -- you'll see the same behavior.

Vincka, 2016-01-18:
Hello,

Ok,if you think that my board is now HS, I just have to buy another one :).

Thank you for your help and thank you for BetaFlight.

borisbstyle, 2016-01-20:
I will close this issue

closed 2016-01-20 by borisbstyle