betaflight/betaflight #112: Add Led transponder system

chinoox, 2016-01-15:
Hi,
I've been studying the transponder systems for my practice and i found EasyRaceLapTimer. https://github.com/polyvision/EasyRaceLapTimer/
This system used a IR led transponder on quad and sensor IR TSOP 31238, all conected to Raspberry PI host station whit Wifi.
The IR Led send binary code to TSOP 31238 and Raspberry tack time and show in a web server.
The think is add IR led code to betaflight and config in CLI quad tag. https://github.com/polyvision/EasyRaceLapTimer/blob/master/Arduino/TransmitterAttiny85v0.2/TransmitterAttiny85v0.2.ino

I do not have time to add the code :(
Sorry for mi english

Trussroads, 2016-01-15:
Sounds interesting.. Nice feature

borisbstyle, 2016-01-16:
This might be usefull for someone in the future to make something work. Thanks

mattyra, 2016-01-20:
Wo hold on this would work !! one-lap do a similar code added to the Attiny85 if you could implement this code out via the ESC pin 5 it would work great......

I have just finished a lap system and was looking at transponder this is a very Viable option lads

YannikW, 2016-01-29:
Hi, I'm the guy from OpenLap. Here is my documentation of the IR Signal : https://github.com/YannikW/Open-Lap/blob/master/docs/Open%20LapTime%20Protocol.md

If someone wanna implement this in Betaflight it would be great :)

But the times are way shorter than normal Looptime so you will have to use one/two timers to do this. Don't know how many timers are left on Betaflight Code..

polyvision, 2016-01-29:
Hey there,

I'm the maintainer of EasyRaceLapTimer. The OpenLap system and EasyRaceLapTimer are using the same protocol (we've worked together on this). If there's a timer left in Betaflight, I could start working on this.

https://github.com/polyvision/EasyRaceLapTimer
nearly everything is documented in the docs directory

konsumkid, 2016-01-30:
I think since everyone who really wanna fly semi pro now is on betaflight, it should be great to have IR Timer support with this as well.. much easier at races..

hydra, 2016-01-30:
SPRacingF3Mini and Cleanflight has transponder support built-in now.

SoulPictures, 2016-01-30:
We know, but we want support for an OpenSource Tracking system such as EasyRaceLapTimer or OpenLap, not only support for this expensive proprietary system.

hydra, 2016-01-30:
@SoulPictures Well, the SPRacingTransponderBEC isn't expensive :smile:

YannikW, 2016-01-30:
But the receiving hardware is... and without documentation of ir protocol it is a closed source system..

SoulPictures, 2016-01-30:
a YannikW you were faster, i was writing the the same. 

GroundStation for OpenLap is around 20-40€ 

The i-Lap system is 140 for the base unit and additional 40 per IR receiver totaling for a comparable system at around 300€ so about 10 times the price...

mattyra, 2016-01-30:
I Agree with Yannik,  the open lap system is the way to go...  Ilap is expensive..  we use it at our club in Kent and it works great, transponders are affordable...

This is a hobby sport and we should make it as affordable for every one...

If Clean flight has the code and support built in then is the Ir number changeable so as not to lap over with existing Ilap transponders all though a slim chance it is possible..

hydra, 2016-01-30:
Noted, I'm gonna bow out of this now.

YannikW, 2016-01-30:
@hydra Why so angry about some suggestion? Transponder on the fc is the way to go I think. So why make only one system possible...

readerror67, 2016-01-30:
Having both would be nice. However, I believe @hydra will offer his via licensing (imagine there is some hash used to generate the output from the ID #). The iLap stuff is way too expensive already, $40 for a crappy little MCU and LED, no thanks. It is however used very frequently and is the standard in the USA anyways.

I think adding a feature that requires extra money to use is the start down a slippery slope.....

readerror67, 2016-01-30:
@polyvision Very cool stuff by the way! I hope this system gets adopted and used more frequently, its an excellent alternative to the legacy hardware/software being used.

polyvision, 2016-01-30:
@readerror67 ERLT gets used a lot ;) Well, not that much than ILap-System but that's because ERLT is pretty young but it works flawless according to all the feedback I'm receiving.

readerror67, 2016-01-30:
It would make sense to re-purpose the LED_STRIP pin/timer/DMA, it should be able to be implemented on every target then without any issues.

jrwrodgers, 2016-01-30:
I'm not an expert in this but given the efforts to reduce processor overhead so that performance is priority then surely having a separate transponder is the way to go. I have recently set up and tested the ERLT (thanks @polyvision :) )  system and it works great with a separate transponder which is easy cheap and light. Just my 2 cents...

mattyra, 2016-01-30:
I like the idea of beta using the open sorce ir seems like a community way to go if possible...

Austen

EdwardDimaguila, 2016-02-03:
This is how the I Lap decoder works.

For the I-Lap hardware there are two known response packets
To start the lap counter you must send the following command
01 25 0D 0A
This initializes the lap counter into 7 digit mode
This will respond with a packet that looks like the following
     01 25 09 32 30 34 09 30 09 30 0D 0A		  
The breakdown is the following
Byte 1 is the header, 01
Byte 2 is the packet type, 25 (%) for an initialization packet
Byte 3 is a seperator character, 09 seperates fields
Bytes 4-6 are the decoder id in ascii, 204 in this case
Byte 7 is a seperator, 09
Byte 8 is not used set to 0 in ascii
Byte 9 is a seperator, 09
Byte 10 is not used set to 0 in ascii
Bytes 11 and 12 are the ending delimiter, 0D 0A is a new line character sequence
		  
After the lap counter is initialized it will send a packet each time a car is detected
An example packet would look like this
       01 40 09 32 30 34 09 33 34 35 36 37 38 39 09 33 32 34 2E 36 32 30 0D 0A
The breakdown is the following
Byte 1 is the header, 01
Byte 2 is the packet type, 40 (@) for a transponder packet
Byte 3 is a seperator character, 09 seperates fields
Bytes 4-6 are the decoder id in ascii, 204 (32 30 34) in this case
Byte 7 is a seperator character, 09
Bytes 8-14 are the transponder id in ascii characters, 3456789 (33 34 35 36 37 38 39)
Byte 15 is a seperator character, 09
Bytes 16 - 22 are the time, it starts counting up when initalized, this field can vary in size, time is in the format SECS.MSS, 324.620 in this case
Bytes 23 and 24 are the ending delimiter, 0D 0A is a new line character
		  
A second example would look like this
   01 40 09 32 30 34 09 33 31 37 34 39 36 36 09 31 32 2E 32 32 36 0D 0A
This has a transponder 3174966 and a time of 12.226
		  
Third example
    01 40 09 32 30 34 09 37 32 35 31 33 31 38 09 34 2E 34 34 38 0D 0A
This has transponder 7251318 and a time of 4.448

YannikW, 2016-02-03:
This is already knowen.. Its the protocoll between Transponder and receiver which is secret..

EdwardDimaguila, 2016-02-03:
Then looking at cleanflight git might be the way to go. since the spf3 already has it. but i still prefer an opensource system that's is why i started my own project based on somebody else's work with some modification. 

Although the timing software is still run on a computer/laptop it manages everything from vtx frequency conflict, transponder conflict, practice runs, qualifying runs, seeding racers to qualifying and finals etc. 

https://www.youtube.com/watch?v=EtEZzbvDjR8

mattyra, 2016-02-03:
Edward... What software are you running for the race timming ??

EdwardDimaguila, 2016-02-03:
@mattyra I am running the latest version of Zround with a custom decoder made from arduino nano and 5 IR receivers for my sensor pole. The decoder box is connected to the PC wirelessly for ease of setup in the field. 

https://www.youtube.com/watch?v=MDhKcGEnzqs

This the main race program but you can also run series championships automatically.
[image]

EdwardDimaguila, 2016-02-03:
I am running the latest version of zround but with a custom decoder built from arduino nano and 5 IR receivers in my sensor post. 

https://www.youtube.com/watch?v=MDhKcGEnzqs

Here what the main race config program looks like. but you can also use the Championship Manager to generate heats and run your races automatically 

[image]

hydra, 2016-02-03:
There certainly are some cool systems out there, I'm looking forwards to seeing what you guys come up.  One thing I know is that I'm glad I updated the Cleanflight recommended hardware to include an F3 with 256KB Flash.  We're gonna be needing it :smile:

Also @YannikW not sure how you thought I was angry, I'm most certainly not.  Quite the opposite.  I'm very happy to have kick-started a revolution in building flight controllers with transponder functionality built in and look forward to what the future holds.

YannikW, 2016-02-03:
Ok, then I misinterpreted this.. Will the source code of the implementation be open? I don't know how iLaps work, but I guess thebprinciple is nearly the same as OpenLap of other projects.. Then implementation will just be a little Modifikation of existing code..

hydra, 2016-02-03:
@YannikW You'd have to chat to the iLaps guys about protocol availability.  I asked them for details but they didn't want to get involved.  An odd move, I thought, since I'm sure they'll sell more kit now!  :smile:
The code transmission system is already open so anyone can build a transmitter using the code.

mattyra, 2016-02-03:
@hydra can you implement the open lap code for the transponder... would make a FC more appealing to clubs who maybe can't invest in the ILAP system

readerror67, 2016-02-03:
@YannikW I am in discussion with someone that has reversed the entire iLap system. I would like to see iLap and OpenLap be an option on every FC, it just makes perfect sense.

Hes getting back to me with some more details tomorrow.

readerror67, 2016-02-03:
@YannikW Oh I forgot, basically what we are missing on iLap is how do they get from the ID on the transponder to the output, I only have 1 (broken... I think) transponder, I believe the number is hashed/xor'd or something in the process before being sent over the air

YannikW, 2016-02-03:
@hydra It is open? :O Can you tell me where to find the specs? I would like to adapt my receiver to OpenLap and iLap protocoll, to make an open source timing hardware with compatibility to both systems :)

@readerror67 Same question to you - where I can find the analysed specs?

readerror67, 2016-02-03:
@YannikW Give me a few and I will get another logic capture done, I had one previously but no idea where the file is.

mattyra, 2016-02-03:
Hows this going guys... Did @readerror67 crack the code :-)

hydra, 2016-02-03:
I can generate iLAP codes Free Of Charge for any non-clone hardware manufacturer or organization that asks for them. The the transmission code can be implemented in other hardware.  No need to reverse engineer or publish the code.

YannikW, 2016-02-03:
@hydra So you have a deal with iLaps, so you know the alogrithm (some kind of ID, XOR and CRC I guess?!) to generate the 6 Bytes of transmission? For any transponder build this is okay, but i would like to make a compatible receiver, so I have to know how to decode the signal..

waslater, 2016-02-03:
I'm a non-programming lay person here, but I'm more concerned about having an open-source solution on the transponder device. I envision a scenario where in the Configurator, I select which lap tracking protocol I'd like the built-in transponder to use.  Might include an option for iLap, Trackmate, Openlap, etc... to match the receiver system I'm using. 

I can understand that a particular manufacturer, such as iLap would want to prevent others from producing a clone receiver if they have some competitive advantage.  But I DO want the ability to use those protocols on my built in transponder if, for example, I go participate in a race with a group who uses a different lap tracking system.  

Is this possible?

readerror67, 2016-02-03:
Well, given there are so many different systems, it makes sense to accommodate all of them, if you are missing one that is frequently used you are shooting yourself in the foot functionality wise. I would like to be able to set my own 'unique' ID to every FC I have and never worry about it again, this is where iLaps model fails. Its kind of a PITA to swap and move transponders around, specially with some of todays super integrated/tight builds.

Openlap sounds awesome, but until ever venue adopts that we are stuck with what we have.

hydra, 2016-02-03:
@waslater Yes, it's possible to add different transponder systems into cleanflight, just as we do for receivers already.

waslater, 2016-02-04:
@hydra That's nice to know. Thanks!

mattyra, 2016-02-04:
Great work lads... We have the Open lap system let me know when the code is implemented would be great to test one of the new SP3 Transponders....

owenson, 2016-02-27:
@EdwardDimaguila Have you decoded the 6 byte packets from the transponders themselves?  I'm presently reversing them, just waiting on some more transponders - maybe you can save me the effort?

owenson, 2016-02-27:
alternatively if someone wants to send me a series of transponder 6 byte codes, I'm fairly experienced reverse engineering this sort of stuff.

EdwardDimaguila, 2016-03-01:
@owenson No i haven't as i don't currently have I-Lap PT's with me.

owenson, 2016-03-01:
@EdwardDimaguila ok.  happy to share my findings, hopefully getting some through this week.  May try and pull the firmware off the avrs.

hydra, 2016-03-01:
@owenson I sent you an email to your gmail account regarding this.

owenson, 2016-03-03:
OK I've almost completely decoded the ilap transponder code pattern now, is there a link to others working on open-sourcing this?

deepimpactio, 2016-03-03:
@owenson they are in this thread https://github.com/YannikW/Open-Lap/blob/master/README.md and https://github.com/polyvision/EasyRaceLapTimer/blob/master/README.md I am not sure if you have looked into these projects

owenson, 2016-03-03:
@YannikW messaged you on facebook, didn't realise you were the same bloke I was chatting to earlier.

slacker87, 2016-03-03:
@owenson https://github.com/slacker87/CoreIR I've been doing some work on it.

hydra, 2016-03-03:
@owenson did yuo get the email i sent?

owenson, 2016-03-04:
Hi all.  The ilap transponder codes are completely cracked, including checksum generation, watch this space for details!

deepimpactio, 2016-03-04:
Wohooo

owenson, 2016-03-04:
Anyone for vanity transponder codes?

3133700 = 0c 36 ec c8 ff 96
0313375 = 0f 44 ce cc 8a 81
1120911 = 0e 96 ed f6 ee 52

owenson, 2016-03-05:
I'm just investigating whether there is another check digit within the transponder code itself (e.g. like credit cards)  On my ilap, the final digit is in a different font which is suggestive.  I've got a fair few codes now,but if anyone can send me anymore please do (either the 7digit or block of 6 numbers is fine).

owenson, 2016-03-05:
@EdwardDimaguila are those transponder codes in your post genuine ones?  Just trying to work out whether the last digit is a check digit or not.

EdwardDimaguila, 2016-03-07:
@owenson I'm not sure if codes are genuine ones, the example is from the documentation of a known opensource laptiming software Flipside Racing.

owenson, 2016-03-08:
Hi folks, I've put a write up of reverse engineering the iLap and the technical details for anyone that is interested.:

http://blog.owenson.me/reversing-ilap-race-transponders/

deepimpactio, 2016-03-08:
@owenson Interesting write up

[21 more replies]

closed 2016-07-03 by borisbstyle