betaflight/betaflight #114: Intermittent arming and oscillation on throttle up

bbbenji, 2016-01-16:
[Issue has been posted to CF](https://github.com/cleanflight/cleanflight/issues/1685) but duplicating here because it affect BF as well.

For some reason at least BF 2.3.0 and 2.3.1 as well as CF 1.11.0 are causing me issues with my Flip32+ v2.3. Arming is intermittent wether I set a switch or use the stock method with thumbstick. Once it actually does arm I get extreme oscillation when I give throttle, it stop when I relax on the throttle. This happens right from start with low values on throttle.

Screenshots: http://imgur.com/a/URLWR
Video: https://youtu.be/0Gb7qQht6x0
Dump: http://pastebin.com/TVAsNVHP

Flashing CF 1.10.0 have resolved these issues for me. Have not tried older versions of BF yet.

carebear482, 2016-01-16:
Why you have "feature RX_PPM" enabled with "set serialrx_provider = SBUS" ?

bbbenji, 2016-01-16:
Good question. It was left over from a previous configuration. Does it make a difference though?

Anyway The problem existed on a clean install as well. Let me verify and post a new dump.

borisbstyle, 2016-01-16:
I think the problem is in your radio and not fligt controller firmware.
Assuming its the real PPM and not something else than PPM signal is not good

closed 2016-01-16 by borisbstyle

bbbenji, 2016-02-08:
@borisbstyle If you follow this issue: https://github.com/cleanflight/cleanflight/issues/1685, you will see that my problem the entire time was with MultiWii (rewrite). Git bisect showed that commit https://github.com/cleanflight/cleanflight/commit/e347e7d4bf28ef15a214ea01643355910e39ce6f is where the problem started.

Unfortunately on betaflight, MultiWii (rewrite) results in chaos. The only other PID Controller I am able to select and successfully apply is LuxFloat, but once LF is selected, I am unable to arm. Where as in CleanFlight I was able to select, apply and arm LF with no issue.

Preliminary indoor tests show that LF in CF may be stable. I will need to do more testing outside.

For now, BF is unable to provide me with even a somewhat stable quad.

borisbstyle, 2016-02-08:
You need to retune your PIDs. Rewrite uses different pids, but most likely rev6 can cause problems too.

bbbenji, 2016-02-09:
It's not only Naze32 Rev6 but also Flip32 v2.3. It is hard to tune if you can't get it up in the air at all. I have tried cutting P and D in half several times together and separately but it did not help at all. Looked to be no different.

I think its time to upgrade to an F3 board.