betaflight/betaflight #116: Disable gyro_lpf

BadSide84, 2016-01-16:
How to disable gyro_lpf to enable 2khz mode on CC3D?

borisbstyle, 2016-01-16:
Hi there. Just this commands on CC3D
set gyro_lpf = OFF
set acc_hardware = 1

closed 2016-01-16 by borisbstyle

BadSide84, 2016-01-16:
Boris this command line doesn't works

set gyro_lpf = OFF

carebear482, 2016-01-16:
Post your dump

BadSide84, 2016-01-16:
# dump

# version
# BetaFlight/CC3D_OPBL 2.3.1 Jan 14 2016 / 16:19:37 (1dddd65)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature ONESHOT125

# map
map AETR1234

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
[... 248 lines cut ...]
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set gyro_lpf_hz = 50
set dterm_lpf_hz = 50
set insane_acro_factor = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 120
set rc_expo = 69
set rc_yaw_expo = 15
set thr_mid = 50
set thr_expo = 0
set roll_rate = 80
set pitch_rate = 75
set yaw_rate = 60
set tpa_rate = 0
set tpa_breakpoint = 1500

#

borisbstyle, 2016-01-16:
in 2.3.1 there is a bug with OPBL version.
You need to flash the hex or use 2.3.0

BadSide84, 2016-01-16:
Ok thanks :)

borisbstyle, 2016-01-17:
you can also use these files if you want the latest version
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/

BadSide84, 2016-01-17:
You mean the bin or hex files? Because i dont want delete the CC3D
bootloader

BadSide84, 2016-01-17:
Thanks Boris, just updated the bin file and works ;)

2016-01-17 9:03 GMT+01:00 BadSide84 FPV <badside84dcft@gmail.com>:

borisbstyle, 2016-01-17:
on that link you can use bin file

BadSide84, 2016-01-17:
Yes, but in 2khz mode cobra 2300 and Kiss18 goes hot i think not good

borisbstyle, 2016-01-17:
not on 2.3.1...only on 2.2.0 version

BadSide84, 2016-01-17:
I mean this last release

BadSide84, 2016-01-17:
I try some hovering for 1-2 min and goes hot!

The D factor is important to not put too high?

carebear482, 2016-01-17:
Looking forward to get also solution for that problem. Boris if you mind, can you send your own dump? Maybe I can track down what is causing this.

borisbstyle, 2016-01-17:
it is already solved in the last source