betaflight/betaflight #119: Motor glitch

Mattawonga, 2016-01-17:
I have glitching problem when Pitch and Roll rate are higher than 0.15

when i go full stick on pitch or roll, glitching will occur and motor/esc gets very hot. got esc go into thermal also.
unless i go full open on stick, quad flies pretty good. but under aggressive flying glitching occurs
 
no problem when pitch/roll rate are at 0.00
i tried luxfloat and rewritere , same problem. crashed 3 times so far

Rg20 latest bl heli
 cobra 2204 / 2300kv
naze 32 rev 6 
2.3.1 betaflight on luxfloat or rewrite.. same problem

i can do a video for better explanation

Mattawonga, 2016-01-19:
https://youtu.be/U0SK1SqXaCc

https://youtu.be/VbHpoePjDJ0

here is the videos

borisbstyle, 2016-01-19:
Is your min_throttle set correctly? If thats too low the motors can be at
the point of stopping. That is exactly why full throttle works.

increase min_throttle by few steps.

Mattawonga, 2016-01-19:
actually, this happened while flying and hard cornering. quad crashed due to this happening. 

can the min throttle create a problem even when flying? i mean i was going pretty fast. did try it 2 times to make sure.
flying slowly i get no issue and quad flies great

borisbstyle, 2016-01-19:
Yes.... min throttle is especially reached in cornering. Lets say you make a sharp right corner....left motors might go full throttle while right motors can reach minimum value.  If min value causes motor to stop you get this issue.

Mattawonga, 2016-01-19:
also, i have 2 tweaker 180, exact same setup. the bl heli might not be the same. 

only difference is the one working good has a purple naze32 rev6

the one in the video has a black naze32 rev6

borisbstyle, 2016-01-19:
yes but min throttle can be different per esc

Mattawonga, 2016-01-19:
okay, do i increase the min throttle in CF for all 4 esc at once or is there a way to feet the sweet spot of each esc?

Mattawonga, 2016-01-19:
""find"" not feet loll

Mattawonga, 2016-01-19:
tried increasing by 5 points at a time, went up too 1180, still having the issues,

should i go higher?

borisbstyle, 2016-01-19:
use 2.3.1 release...

there is an issue in 2.3.2 regarding this

Mattawonga, 2016-01-19:
i was on 2.3.1 before, and had this issue, went back to 2.1.6 issues still there... i can flash again to 2.3.1 if needed

but after more inspections, i dont need to be aggressive on the left stick for issue to happen! seems that about 90% of stick input will toggle the problem... at 80% and 100%... no problem,

take a look at this video

https://youtu.be/yw8n5LSFo6M

borisbstyle, 2016-01-19:
In your case its something else. Def your gear as nobody else has this on pre 2.3.2 versions

closed 2016-01-20 by borisbstyle