betaflight/betaflight #121: Build number or timestamp for releases

mikeller, 2016-01-18:
Hi Boris.

Is it possible to get a build number or a timestamp added to the releases? Keeping track of multiple versions being released with identical version numbers is a bit tedious and confusing.

Also, would it be possible to have the version / build numbers added to the names of the hex files for download?

deepimpactio, 2016-01-18:
There is build number on release page under version which is currently 2f6eeed. It would be nice if upon downloading it would append version number. I rename and do it myself as a workaround

pqueiros, 2016-01-18:
Maybe there is a good reason for not doing it this way. But why not just increment the version?eg. after major release 2.3.1 we could have 2.3.1.1,  2.3.99 etc. 

Just a thought. ;)

mikeller, 2016-01-18:
Hi deepimpactio.

On 19 Jan 2016 1:42 am, "deepimpactio" <notifications@github.com> wrote:
>
> There is build number on release page under version which is currently
2f6eeed. It would be nice if upon downloading it would append version
number. I rename and do it myself as a workaround

I think you are confusing the changeset id with a build number.

mikeller, 2016-01-18:
On 19 Jan 2016 1:45 am, "pqueiros" <notifications@github.com> wrote:
>
> Maybe there is a good reason for not doing it this way. But why not just
increment the version?eg. after major release 2.3.1 we could have 2.3.1.1,
2.3.99 etc.

Any way that preserves the history and allows reviewing of release notes
for any particular release on order to figure out why something worked one
way or another would be appreciated.

deepimpactio, 2016-01-18:
I thought the commit number was utilized as the  build number,  but I could most definitely be wrong.

borisbstyle, 2016-01-18:
commit number is indeed always utilised into the build and release

mikeller, 2016-01-18:
Hi Boris.

It looks like for some reason github is not properly recognising the build number in the way that it is set on your releases. It seems to overwrite release note for existing releases every time you publish a new release with the same version and a new build number. For example, I am unable to find the release notes for version 2.3.2-2f6eeed, which is what is installed on one of my FCs.

borisbstyle, 2016-01-18:
@mikeller 
The release notes are same....the first version I quicky released had an issue with motor stop....it was unintentionally broken. Thats why I removed it to prevent confusions

mikeller, 2016-01-18:
Hi Boris.

I think it is more confusing if there are versions out there and being used by people, with no history available for them, and no way to tell what was released and when. I.e., in the current situation it would not be surprising if people reported the issue with motor stop against 2.3.2, simply because there is no way to tell that there is a version that has been released that has not had this issue fixed.

borisbstyle, 2016-01-18:
@mikeller 
Remember this is betaflight ;). The main reason it is called like that is the fast development what requires some public testing and feedback. That was my initial intention.
.....and easiest way to do testing is to release something in this case. 2.3.2 is a special build for me as I am doing something with 3D without ability to test it, and I get many requests for that. 

I did make sure that users without 3D mode would not be affected at all to be able to do future changes in 3D code. Some rework in mixer was required for that.

There is a very easy way to see the version and that's the github commit number visible in cli when typing version or doing the dump.

mikeller, 2016-01-18:
Hi Boris.

Fair enough, I totally understand that one of the goals of betaflight is to put out new versions for testing fast. But even more so, I do not think there is a need to 'pretty up' the release history by removing superseded released versions. Having a clear history of what has been done and when would be helpful.

In addition to seeing the version 'number', it would be helpful to see what has actually changed between this version and other versions that have been released, in order to be able to decide if it has no effect / makes sense / is necessary to flash a newer version (there might not be any changes for the board in question, or all of the changes might be for features that are not used in the configuration in question, or it might be a bug making a crash likely).

borisbstyle, 2016-01-18:
@mikeller 
you can track all changes here 
https://github.com/borisbstyle/betaflight/commits/betaflight

mikeller, 2016-01-18:
Hi Boris.

That's the commit messages. These can be quite a bit of work to read through, especially in the case of merges from upstream. I think most users would be better served if the original release messages were preserved, so they could get an abridged overview of the main changes between the version they are using and different versions.

readerror67, 2016-01-18:
@borisbstyle 

It may be worth looking in to this:
http://stackoverflow.com/questions/1704907/how-can-i-get-my-c-code-to-automatically-print-out-its-git-version-hash

If its added in to the code its nice because you can look at the commit hash and find exactly the point it was when compiled.

Maybe worth adding to the CLI 'version' output.

borisbstyle, 2016-01-18:
It is there in case you didnt know!

Short git version is reported in the dump or when typing version ;)

borisbstyle, 2016-01-18:
I will keep it cleaner next time guys. I usually do that, but this initial 3D mode seriously needs testing. I dont want to mess a lot with spread test binaries all over the place.

Sorry for confusion. 2.3.2 is now released from pre-release btw. Same binary. Will wait for feedback about 3D and if there are issues I will release another patch for that.

borisbstyle, 2016-01-18:
@readerror67 
I still like the short git version. I think it's fine. Github releases offer a good way of managing versions.

closed 2016-01-18 by borisbstyle

mikeller, 2016-01-18:
Thanks, Boris.

If you are interested in getting specific things tested, what about adding a list of 'test cases' to your release notes, and get users to report back on these aspects?

borisbstyle, 2016-01-18:
@mikeller 
Yes I added tht to the current version that I need 3D test cases. Already had some valuable feedback