betaflight/betaflight #122: jitter since version 2.3.0

raymonkroeze, 2016-01-18:
Hello,
Last weekend i installed the 2.3.0 firmware. but since then i had some jitter on my quad. at first i thought i had the Pgain to high, but this is not the case. I also tried 2.3.1 this afternoon and im still having jitter. When i arm the quad i can hear the motors are spinning not as smooth as usual. 
Also with airmode on and 0% throttle sometimes the quad feels like it just falls to one side and lose some control.
Then i installed version 2.2.0 and the problem is gone.

Im flying with luxfloat and have not tried rewrite on this one.
My setup:
Naze32 rev5 acro
little bee 20A esc with blheli 14.3
emax 2204 2300kv motors
180 size frame 
HQ4.4.3 props

To me it's not that much of an issue because i really like how 2.2.0 feels. I just wanted to let you know there might be an issue.

Overall im very very very happy with betaflight!!!!

borisbstyle, 2016-01-18:
try lowering D

Rob922, 2016-01-18:
I have the exact same problem Raymon. On 2.2.0 it flew close to perfect but now with 2.3.2 it is jittering like crazy on punchouts. I had to use TPA to reduce it but I never had to before. I never had to upgrade the firmware when it flew good.

But a friend of mine with a similar setup had no problems with 2.3.2 whatsoever. I think it is because I have more vibrations from my my motors compared to him.

I think it has something to do with the dterm_cut_hz defaults being changed.

Version 2.2.0:
"New defaults. Note that dterm_cut_hz is disabled by default now due to better moving average. It still can be enabled"

Version 2.3.1:
"Added Configurable soft filters set gyro_lpf_hz = 0-255 (50 default, 0 means disabled) set dterm_lpf_hz = 0-255 (50 default, 0 means disabled)"

Maybe by disabling the dterm_lpf_hz it will fly like in 2.2.0 ?

borisbstyle, 2016-01-18:
no it wont...

disabling will make things wors. Just lower D gain.
You can also lower filter values if your setup is noisy

Rob922, 2016-01-18:
Oke will try that. thanks for the response!

raymonkroeze, 2016-01-18:
I will try the. I had my D-gain at 20.

Mattawonga, 2016-01-19:
same problem on 2.3.1 and luxfloat. tried rewrite, no jitter

flashed back to 2.1.6 on luxfloat no more problem

borisbstyle, 2016-01-19:
Try lowering D

D works differently and needs retune

Mattawonga, 2016-01-19:
will do that! thanks

borisbstyle, 2016-01-20:
try 2.3.3a

closed 2016-01-20 by borisbstyle