betaflight/betaflight #126: More inflight adjustments.

jaimevisser, 2016-01-20:
Currently use inflight adjustments for rough tuning but it is limited to 11 slots while there are 20 functions. If I want to adjust P,I,D,rate per axis and rc rate/expo I need 14 slots. Missing 3 adjustments is a royal pain.

pqueiros, 2016-01-20:
Hi Jaime, that is a valid point. But what I find myself is missing enough physical switches for that. How do you do it? I guess I am not taking full capability of my radios (hitec aurorax and taranis).

jaimevisser, 2016-01-20:
@pqueiros:

With Taranis:
- Disable all trim buttons from flight modes screen.
- Set up one trim (I use roll trim) to do +/- on a gvar.
- Create an aux channel based on this gvar, spreading out about 14 gvar values across the channel band.
- Let you Taranis speak out the gvar when you change it (really helpfull)

- Create Logical switches for trim (I use pitch trim) up / down.
- Make aux channel that behaves like a momentary switch using the Logical switches

In ~~clean~~betaflight
Set up the inflight adjustments to use the first channel to select the adjustment you want to use and the second channel to do the actual adjustments.

While flying switch to the adjustment you want with roll trim, then change it with pitch trim. After landing and disarming either save with stick command or disconnect the lipo to forget changes.

I also requested the coder of beta-osd to add inflight adjustments to his osd software so we can see what we are doing.

LeChuck71, 2016-01-20:
Almost same here on a 9xr

I use a "poti" knob to select the function spreaded over the band (currently just PID) and the 3pos switch to do up/center/down
On "Switch on" check i make sure that the 3 pos is centered so I accidentally do not change any values. Maybe its a good habit to use a different model on your tx to do adjustment work

pqueiros, 2016-01-20:
@jaimevisser and @LeChuck71 

Thank you very much for the feedback. Looks really promissing :+1:

chasers, 2016-01-20:
Pretty n00b here, but I set up inflight adjustments on my Taranis. Got it all working just fine, not quite ideal though.

What I thought would be super easy would be to have one knob select roll / pitch / yaw (depending on position) and the other knob select p / i / d (depending on position), and then you have your 3 position switch acting like it does already moving the value up or down.  

Then any of the other 3 pos switches can act on rc rate and rc expo.

idk, maybe this is possible already but I didn't have too much time to experiment.

lacymorrow, 2016-01-20:
@chasers definitely possible, if you follow a guide to create a 6-position channel on your taranis you can map out r/p/y and p/i/d to those 6 points.

KiteAnton, 2016-01-20:
I don't know if it is some limitation or if it will work on every target but there is a define parameter, MAX_ADJUSTMENT_RANGE_COUNT, that is set to 12 in the rc_controls.h. I guess increasing this value will allow more adjustment possibilities.

KiteAnton, 2016-01-20:
The limited amount of flash for some targets (like Naze32) will limit the possibility to increase the number of adjustments. I made a proposal for additional adjustments, it is here https://github.com/KiteAnton/betaflight/tree/additional_adjustments 
I don't know the actual demand for this but you if anyone wants it you should make a pull request for this to betaflight .

jaimevisser, 2016-01-21:
Nice @KiteAnton! I do hope that @borisbstyle will be able to merge this.

Maybe we need to simplify adjustments to only use two set channels (one for setting the adjustment type and one for up/down) so we do not need to store the channel for every adjustment. That would decrease flash needed for the storage of adjustments a bit, making more adjustments an easier fit. This would also imply that MSP and configurator need to be changed to reflect this.

KiteAnton, 2016-01-22:
I guess also reducing the number of profiles will free up some flash. I don't know how many people are actually using 3 profiles? Maybe 2 will be sufficient ?

jaimevisser, 2016-01-22:
To be honest, I just use one. Will poll at MRO (my local quad fanatics group, ~25 members) how many they actually use.

jaimevisser, 2016-01-22:
Most use one profile, 2/25 use two profiles during tuning sessions.

jaimevisser, 2016-01-22:
Also; to get an example of setting up trim switches to do inflight adjustments see:

https://slack-files.com/T0HT65BAS-F0K33E6J0-4d02e18df8

This is an epee file for openTX companion to show an example of how you can combine a lot of functions on one channel and use the trim buttons as mode or 3-way momentary switches.

Aux1 combines:
SF is disarmed/armed/air-mode switch.
SE is buzzer, turning it on also disarms the quad. This is meant only for finding a crashed quad.
Roll trim changes the inflight adjustment by doing small increments. Only works in air-mode.

Aux2:
Pitch trim for up/down adjustments.

Also check beta-osd for minimosd. It might soon add support for inflight adjustments to the osd.
https://github.com/basdelfos/beta-osd/issues/1

KiteAnton, 2016-01-26:
I made some tests to rearrange config-data and managed (for naze target) to free up 120 byte flash and 540 byte RAM. But it is done with some sacrifices:
* Modes is moved from profile to master
* Adjustments is moved from profile to master
* Profiles is reduced from 3 to 2

Is there a great demand for in-flight adjustments and modes to be different for each profile ? With this change both modes and adjustments will use same settings for all profiles. When I previously used profiles I would say this is better so you don't have to set up everything twice. 

I have this ready as a pull request, I can file it later but I'm not sure everyone agrees that this sacrifice is desired.. ?

jaimevisser, 2016-01-26:
This is great as far as I am concerned as it will keep profiles for switching pids and keep the rest of my settings the same.

jaapp, 2016-01-26:
+1 - no need to have different adjustment settings for different profiles.

borisbstyle, 2016-01-26:
Yeah those profiles are awesome. Almost everything has to be reconfigured
and it all takes 3 times more flash space :D. Whoohoo!

mikeller, 2016-01-26:
I totally agree with @jaimevisser: Profiles make good sense for switching rates and PIDs. Just about every other setting that is contained in the existing profiles is tied to an existing hardware configuration, and short of wanting to be able to swap an FC between different craft, there is little use for changing it when changing PIDs or rates. (To get around the tedious reconfiguration for different profiles, http://www.etheli.com/CF/clipytool/ allows dumping of the config, editing the text file, and then uploading it again.)

borisbstyle, 2016-01-26:
thats not the point cause when you want to change one parameter like 1 mode
or one aux channel you have to do it 3 times.

mikeller, 2016-01-26:
Hi @borisbstyle 

> thats not the point cause when you want to change one parameter like 1 mode
> or one aux channel you have to do it 3 times.

I don't quite understand. What part of my statement does this refer to?

borisbstyle, 2016-01-26:
not your statement....i am saying that i found the current profile
structure annoying.

Trussroads, 2016-01-27:
I only use a few slots at a time usually PID on pitch roll yaw and find its no big deal to plug in and reload those with other parameters.. It keeps the tuning physically simple with the TX (after all, how much can you do on one battery anyway?) So personally the number of slots and range limit don't worry me.. I couldn't be bothered spending so much effort cramming functions into the radio when a few clicks on PC and I'm sorted anyway?
I agree with the rest of this though, Keep it simple Sir ;D

Regards,
Scott Chard..
0448411827.

fedorcomander, 2016-01-27:
Btw, i replaced poti on taranis with 6 position encoder ;))) Very happy with it in the area of inflight adjustments. It has 6 positions which i multiplied with "shift" switch to 12 using mixes. 

U use this one, just drop in replacement.

http://flyduino.net/Taranis-6-position-flight-mode-switch-for-APM/Pixhawk

Enjoy.

timdonovanuk, 2016-02-05:
+1 to sharing modes between profiles. This caught me off guard in the past where I wanted to try a different PID controller so set a different one to a different profile..forgetting that I also need to re-setup all the modes in the new profile too. I took off expecting to flick into Angle mode (it was very untuned and wanted stability to land) but was stuck in Rate mode!

jaimevisser, 2016-02-05:
@borisbstyle, @KiteAnton: Seems like a fat majority can live with two profiles and actually likes modes and other stuff moved into master. Do you agree this would fit into betaflight or should I just build it separately if I want it that badly? ;)

borisbstyle, 2016-02-05:
@jaimevisser 
There is a PR waiting for that. I just asked the question on RCG to see how many people will not be happy about it. I personally don't mind, but need to check things like compatibility with configurator etc.

borisbstyle, 2016-02-05:
This is the PR

borisbstyle, 2016-02-08:
closing this one. Added in 2.4.0-RC1

closed 2016-02-08 by borisbstyle