betaflight/betaflight #127: CC3D - Change PPM input pin

henn1001, 2016-01-20:
Summary: http://www.rcgroups.com/forums/showthread.php?t=2464844&page=1014#post33773011 and following posts

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L636
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L147
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/timer.c#L70

I never liked that hack and it seems to be causing issues now. The ppm should be moved to a pin with either Tim2 or Tim3 to avoid this clash. I suggest the rssi pin / same pin Openpilot is using to avoid this conflict.

Maybe someone can tackle this cause i don´t have the time at the moment.

borisbstyle, 2016-01-20:
Thanks for pinpointing the issue! That might eventually calm down those CC3D complaints :D

borisbstyle, 2016-01-20:
@henn1001 
I still need help on this one as I can't test. I blew up my only CC3D board a while ago :(

Cwix, 2016-01-20:
I'm willing to help if possible.

borisbstyle, 2016-01-20:
@Cwix 
Feel free to add a PR for this!

Cwix, 2016-01-20:
@borisbstyle Hah, yeah right, don't think my limited embedded knowledge goes that far. But I will take a look, who knows.

borisbstyle, 2016-01-20:
From coding perspective it should not be that difficult. You do have to
test and know hardware.

Its good for learning ;)

Cwix, 2016-01-20:
Yeah, I'll grab the datasheet. Haven't touched ARM before to be honest, but as you said, its good for learning :-)

henn1001, 2016-01-20:
It should just be a matter of removing this https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L635

Change this to PWM6 https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L148
(it relates to this timer count here https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/timer.c#L70 - count upwards to map pwm to a timer)

And prevent the usage of ADC rssi. That should mostly cover it but ofc there could be other code changes necessary.

borisbstyle, 2016-02-08:
This issue is resolved in this PR. It will be available in 2.4.0 RC4. Thanks to @edusteinhorst
https://github.com/borisbstyle/betaflight/pull/170

closed 2016-02-08 by borisbstyle