betaflight/betaflight #129: Motors still get hot in 2.3.3

carebear482, 2016-01-21:
As the title says, simple hover test with 4045 normal shape prop with 2633 kv motor on 4s will make motors kind of hot, can't imagine what would happen with full throttle. Any ideas?

borisbstyle, 2016-01-21:
Probably overtuned D combined with noise of the frame.
Blackbox would be handy. Usually lower D would help

But post your dump first.

As the title says, simple hover test with 4045 normal shape prop with 2633
kv motor on 4s will make motors kind of hot, can't imagine what would
happen with full throttle Any ideas?

carebear482, 2016-01-21:
I messed with blackbox and bricked my naze, can't give you that dump. But it was default with sbus,telemetry enabled. I reflashed with 2.1.6, tested again and motors are cold with high timing. You should really investigate this issue and compare with 2.1.6.

borisbstyle, 2016-01-21:
There is actually no issue at all.

2.1.6 was 1khz.....put 2.3.3 to 1khz and it will be the same.
If you go for 2khz than you should retune....usually low D will help.
You dont need more D than 10

carebear482, 2016-01-21:
Still no difference. Reflashed 2.3.3 to test that.

borisbstyle, 2016-01-21:
retune it. Set D to 0 and slowly increase in steps

carebear482, 2016-01-23:
I tried to spin motors on motors tab and even then motors are heating up

borisbstyle, 2016-01-23:
than you have an issue with your gear :)

carebear482, 2016-01-23:
How come everything is okay in 2.1.6 then?

readerror67, 2016-01-23:
Man, idk....

Github issues should be used for code related things, not general support. I am signed up for notification and too many people seem to use this as a helpdesk. But yea this seems unrelated to the code given it happens in tester.

Now where is my close issue button...

carebear482, 2016-01-23:
Well if I am really the only one with this issue what truly seems to be related with code then oh well, I'll close it.

closed 2016-01-23 by carebear482