betaflight/betaflight #132: Add support for binding R616XN receiver

sjtrny, 2016-01-22:
The R616XN receiver is a nano PPM receiver. To put it into bind mode you set or connect the signal pin to ground. However this is a huge pain if you have the R616XN direct soldered to the FC and buried in a small quad.

This issue is a feature request for adding a small CLI utility to assist with binding. It would be used as follows:

    set ppm_bind = 1

This will set a variable `ppm_bind` in the `master_t` struct which is stored on the EEPROM.

Next the user will reboot. Upon reboot the FC will check if `ppm_bind` is enabled. If so it will set the corresponding PPM pin for the target to GND. This has to happen fast.

Once the bind process is complete the user will then set

    set ppm_bind = 0

Which sets the variable to 0 and normal operation is resumed.

quiljw01, 2016-01-25:
I have a problem with this version of betaflight, version 2.3.0.  Hardware I'm using Afroflight Naze32 rev6 with Orange r616xn receiver plugged into the gnd, +5v and PPM pins on the flight controller.  Nearly every time I power up the flight controller the receiver is put into bind mode. It seems like about 5 or 10% of the time it will power up normally (not in bind mode) but i don't know whats causing it.  I dont' know what else to rule out because the same exact receiver was working fine when it was on my cc3d/ openpilot board.   help?!

borisbstyle, 2016-01-29:
@quiljw01 
That doesnt sound like a problem in software. Is that PPM?

quiljw01, 2016-01-29:
Yes it i think it could very well be a problem with the FC but I don't know how to test.  Could probably put a 2 channel scope on the 5v and PPM line but I'm not sure how to interpret the results.  

Yes r616xn is a very small and inexpensive PPM receiver.  This is not an urgent problem for me any longer as I have switched to a spectrum satellite Rx which is working well.  Its just unfortunate because the r616xn is so small.  

http://www.hobbyking.com/mobile/viewproduct.asp?idproduct=73005

martinbudden, 2016-06-17:
Closing, as marked as 'wontfix'.

closed 2016-06-17 by martinbudden