betaflight/betaflight #133: Buzzer Issue

timmyjane, 2016-01-23:
I can not get the buzzer to work on motor pin 6.  CC3D board in ER250.  Is it confirmed to be working?  Buzzer is enabled in CLI.

KevinVecchione, 2016-01-25:
Are you using this circuit? https://github.com/cleanflight/cleanflight/blob/master/docs/Wiring/CC3D%20-%20buzzer%20circuit.pdf

timmyjane, 2016-01-25:
Yes.  The circuit will work with Arduino but not with the CC3d. Since posting the issue I did confirm that the Pin 6 was outputting signal and would make the active buzzer work when bypassing the resistor and transistor so this is not a Betaflight issue.  Any help on the circuitry would be great/

timmyjane, 2016-01-25:
I am going to close this issue as I have confirmed BF is working.  My problem is documented here:  http://www.rcgroups.com/forums/showthread.php?t=2583742

closed 2016-01-25 by timmyjane