betaflight/betaflight #135: Compass problem with MPU9250

gamani, 2016-01-23:
Hi Boris,

First thanks for your amazing work !

We are designing a new board and we need your help.

The Sp racing F3 used the MPU6050, compass is 5833.
For our new board we use the MPU9250=MPU6500+AK8963 ,
Still I2C as communication  

Now for the MPU9250, we modified the drive firmware,it works well.
But for the compass, we still used the drive firmware of AK8975+HMC5883 (the one for SP racing F3.
We don't know how to deal with it,so it works not good for the compass part.

Any help or advice is welcome.

Thanks
Alex

henn1001, 2016-01-23:
Have a look @raceflight  https://github.com/rs2k/raceflight

https://github.com/rs2k/raceflight/pull/33

gamani, 2016-01-23:
OK will check it thanks

MJ666, 2016-01-26:
There is also a pending PR for Cleanflight wich is adding the AK8963 driver code: 

https://github.com/cleanflight/cleanflight/pull/1638

The problem seen here is the AK8963 is connected at the Internal I2C bus and the SPI communication speed is not really leveraged if we reading the MAG. 

Our plans for the AlienFlight F4 FC is to likely go with an MPU6500 and an separate SPI connected HMC5983 Mag.

For the F3 the driver is working but will likely slow down the cycle time. This is may be not what we want for BetaFlight. At the F4 the driver is working less well. The required delays to make it work will generate quite some jitter in cycle time in any case.

borisbstyle, 2016-01-26:
@MJ666 
Does that really need SPI? Baro/mag update is not that often needed and i2c is more than fine for those.

MJ666, 2016-01-27:
Since I2C and SPI drivers are blocking in all XXXXFlight forks the slower I2C communication is slowing down all other processing and may potentially increases the jitter. May be not an big issue for racing since the additional sensors are barely used but likely interesting for advance navigation features to not waste CPU with waiting for the slow communication. The best way would to use DMA in the driver and interrupts if it is done. I have an Autoquad M4. This has all SPI sensors may be I will try to make an RaceFlight taget for this. Alienflight F4 with all sensors will still take some more time. We actually have some brushed and brushless prototypes for racing.

borisbstyle, 2016-01-27:
Yes DMA would be the way to go!

That would also increase the performance of the current i2c targets.

Just let DMA handle the read action and pick it up once complete.
Would be a major performance bump

borisbstyle, 2016-01-29:
This is fixed in the alienwii commits I believe

closed 2016-01-29 by borisbstyle