betaflight/betaflight #136: Brushed motor support is broken since a couple of weeks.

MJ666, 2016-01-26:
BetaFlight  actually has an bug wich cause brushed motors to spin at power on. It is here in the code:

https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/main.c#L266-L267

It I s fixed in RaceFlight already:

https://github.com/rs2k/raceflight/blob/raceflight/src/main/main.c#L287-301

Her is the same part at CleanFlight:

https://github.com/cleanflight/cleanflight/blob/master/src/main/main.c#L263-L264

Sorry for not making an PR but I have problems to have the CleanFligth and BetaFlight fork at my Git in parallel. This is not really working. Looks to be Git is not see this as an independent fork.

borisbstyle, 2016-01-26:
@MJ666 
I think the scheduler merge broke that. It was in there added by me.

borisbstyle, 2016-01-26:
Fixed here
https://github.com/borisbstyle/betaflight/commit/17f214d2abbb4559b2b4f5f0ea6ad7b101ef500e

closed 2016-01-26 by borisbstyle

MJ666, 2016-01-26:
Cool, thanks.