betaflight/betaflight #139: One Motor Cuts Out When Flipping

colewichman, 2016-01-26:
I looked at the other posts with the motor cut out, but this is different. I can't tell if this is the motor/esc or the betaflight firmware. It takes off fine and fly's around, but whenever I do a flip of some sort, it freaks out and crashes. It just kind of buzzes around like a bee. It has done this three separate times, and one time it left me with a completely broken frame. It looks as if only one or two motors go out, but I'm not positive. I am using cobra 2204/23 and The Beatles 30a ESCs. Please help.

orefalo, 2016-01-27:
looking at the number of people doing flips without issues, most likely a bad solder and an issue with esc ranges. are they blheli? are all configured the same on blheli configurator?

borisbstyle, 2016-01-27:
Yes the issues here are mostly to keep a track of software problems.

This sounds like a hardware problem. Check soldering. Check min_throttle value.

closed 2016-01-27 by borisbstyle