betaflight/betaflight #142: Motors jittering and spooling up strangely

brainmatterfpv, 2016-01-27:
Hi, 

I know its fairly normal for the motors to spool up with airmode enabled.. however mine are doing something very different (they weren't doing this with a previous version of betaflight) 

This video (not mine) is exactly what mine are doing whenever I enable airmode.. causes my motors to get very hot as well. 

https://www.youtube.com/watch?v=eZ37S8MTqGE

Need some help please!!

Thanks

borisbstyle, 2016-01-27:
please try 2.3.4 first

I see in the video it is 2.3.0

closed 2016-01-27 by borisbstyle

brainmatterfpv, 2016-01-27:
Hi Boris, 

I have just flashed my own to 2.3.4 tonight and the same issue is happening. 
I swapped my FC out from a Naze rev 6 to a Dodo to see if it was the issue and its still doing the same thing.. 

Thanks,

jjwilson4, 2016-03-08:
You ever get to the bottom of this?  Having the same issue - twitchy motors that get real hot.

brainmatterfpv, 2016-03-08:
I found that tuning my pids down a lot made a big difference. Still have a few issues here and there though.

Went from 4.5 on Pitch P on previous versions down to 3.0 now.. Big change.